modm-io/modm #316: STM32 ADC channels >= 10 broken

mikewolfram, 2019-11-20:
Commit 086581565e088c4fda60e8d13a8a8b1fc9f0912b (adding support for L1) breaks ADC channels >= 10. A change in *src/modm/platform/gpio/stm32/pin.hpp.in* sets the ADC channel to always 1 when the Signal is `In10...In15`. Before supplying a fix I wanted to ask if this change has a purpose:

```diff
- Gpio{{ port ~ pin }}::AdcChannel<Peripheral::{{ signal.driver }}> = {{ signal.name.replace("In", "") }};
+ Gpio{{ port ~ pin }}::AdcChannel<Peripheral::{{ signal.driver }}> = {{ signal.name.replace("In", "")[0] }};
```

salkinium, 2019-11-20:
Ah, dang. I wanted to filter out only the number, without any suffix.
https://github.com/modm-io/modm-devices/blob/develop/devices/stm32/stm32l1-51_52_62-c_d_e.xml#L498

This should be done in Python via a custom Jinja filter.

salkinium, 2019-11-20:
Maybe something as simple as `"".join(filter(str.isdigit, name))` as a filter would be enough.

mikewolfram, 2019-11-20:
Not sure whether I understand... Use it in the pin.hpp.in or module.lb file?

salkinium, 2019-11-20:
Both ;-P

You'd define the filter in [the `module.lb` here](https://github.com/modm-io/modm/blob/develop/src/modm/platform/gpio/stm32/module.lb#L263):
```py
        def name2adc(name):
            return "".join(filter(str.isdigit, name))
        env.template(gpio_source, header_name, filter={"to_adc_channel": name2adc})
```

Then use this filter in the Jinja template:
```
Gpio{{ port ~ pin }}::AdcChannel<Peripheral::{{ signal.driver }}> = {{ signal.name | to_adc_channel }};
```

salkinium, 2019-11-20:
You can even use an inline Python lambda function instead of the lame `name2adc`.

PR #317 (2019-11-24): Add filter to extract the ADC channel number only
I'm not a Python expert but hope the fix is ok.

Fixes #316.
closed 2019-11-24 by salkinium