modm-io/modm #319: Reproducible builds

salkinium, 2019-12-18:
Compiling an application should reproducible regardless of project path etc.

See https://interrupt.memfault.com/blog/reproducible-firmware-builds#fdebug-prefix-map
See https://blog.galowicz.de/2016/02/20/short_file_macro/

W-M-D, 2020-03-11:
If anyone else is needing something like this I highly recommend that they make a vagrant build and back it up everywhere. Machines may die, projects change, but that virtual machine will always be there.

salkinium, 2020-03-11:
It just would be nice to have two machines generate the same ELF file, for purposes of archiving (via `scons artifact`) and sharing debug information across machines.

W-M-D, 2020-03-13:
Wouldn't you have to have the entire toolchain as well if you do this ?

W-M-D, 2020-03-13:
Not 1 hour after I commented that I did an upgrade of my toolchain that was missing the  cstddef package lol.

salkinium, 2020-03-13:
> Wouldn't you have to have the entire toolchain as well if you do this ?

Yes, outside of having a dedicated build machine (like a CI or Jenkins), there is no guarantee that this will work.

However, we recommend [using the official pre-compiled binaries from ARM](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm), as they *should* theoretically create an identical binary regardless of what OS compiles them. At least they all compile approximately the same source code 😝…
If the builds differ the GNU BuildID will also be different, so at least it's detectable.

closed 2021-09-29 by salkinium