modm-io/modm #326: C++20

rleh, 2020-02-10:
A lot of the C++20 features are already available with GCC9, most big features ([Coroutines](https://en.cppreference.com/w/cpp/language/coroutines), [Concepts](https://en.cppreference.com/w/cpp/language/constraints), ...)  will require GCC10.

For a complete list see https://en.cppreference.com/w/cpp/compiler_support.

salkinium, 2020-02-12:
In addition, the [GCC AVR backend will be deprecated in v10 and removed in v11 (~2021 release)](https://gcc.gnu.org/ml/gcc-patches/2019-09/msg01256.html), so we may gain C++20 while loosing AVR support. Since Cortex-M development is de-prioritized by ARM for GCC as well with new features not being added (Helium extension etc), it may be necessary to switch to LLVM as our primary supported compiler.

henrikssn, 2020-09-11:
What is the main effort of upgrading to GCC 10?

As @salkinium says, AVR is supported (but deprecated) there so that should not technically be a blocker.

salkinium, 2020-09-11:
> What is the main effort of upgrading to GCC 10?

- hosted gcc is v10 in CI and Linux and macOS.
- avr-gcc is v10 in the CI (and Linux) and on macOS. Don't care about AVR+Windows for now.
- arm-none-eabi-gcc is still v9 though and will probably only be upgraded by ARM at the end of the year? Or maybe perhaps already in Q3?

So I'm waiting on ARM.

rleh, 2020-09-14:
Main motivation for the upgrade to GCC10 are C++20 features, mainly [Concepts](https://en.cppreference.com/w/cpp/language/constraints). See discussions #341 and #353.

rleh, 2020-09-14:
> hosted gcc is v10 in CI and Linux and macOS.

Most Linux distributions (Ubuntu, Fedora, Arch, ~Debian~) have GCC10, that probably concerns most modm users.
MinGW is only version 9.2, so Windows CI would be an issue. (Hosted on Windows could be disabled in CI until MinGW update is available.)

> avr-gcc

I hope AVR-support will stay in GCC after version 11, there is a chance: https://www.bountysource.com/issues/84630749-avr-convert-the-backend-to-mode_cc-so-it-can-be-kept-in-future-releases

rleh, 2020-09-14:
> So I'm waiting on ARM.

There already is a ARM GCC 10 "Preview": https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads#panel2a

salkinium, 2020-09-14:
Hui! So I can already play around with Coroutines on Cortex-M? The fibers are quivering!1!!

salkinium, 2020-12-18:
ARM GCC 10 is now out of preview and available!

rleh, 2020-12-18:
Nice!
I'm going to update our CI docker images (and move them to Github Container Registry ghcr.io) and update the installation guide, ok?

salkinium, 2020-12-18:
Yes, please!
The macOS CI will update itself once my PR is merged: https://github.com/osx-cross/homebrew-arm/pull/16

salkinium, 2020-12-19:
Looking at the [GCC Standards Support](https://gcc.gnu.org/projects/cxx-status.html) we still won't get full C++20 features with v10 and even in v11 some things are still missing.

So we have to be a little careful about what features we use.

jasa, 2021-01-27:
Does this have interest [C++20 Register abstraction](https://github.com/jasa/regist/blob/master/README.md)?

jevinskie, 2021-05-31:
Good news, the AVR CC0 port bounty was completed and the work was checked in to mainline.

salkinium, 2021-05-31:
avr-gcc v11 is also available on macOS and we can port the changes (if any) to our avr-gcc Linux repo:
https://github.com/osx-cross/homebrew-avr/commit/252b2f5aa7a48ba2e1e50fbec33a5c524fb4d935

closed 2021-09-29 by salkinium