modm-io/modm #327: ST7565 display: "initialize" never called?

WasabiFan, 2020-02-20:
I was referencing the ST7565 implementation when writing my own display controller driver. It appears that the protected `initialize` method isn't ever called by a parent class, and can never be called by an external user since it's `protected`. What am I missing here?

https://github.com/modm-io/modm/blob/develop/src/modm/driver/display/st7565.hpp#L51

salkinium, 2020-02-21:
The driver is just really, really old and has probably just not been ported correctly by me. I think this is just a bug.

salkinium, 2020-03-13:
Has your question been answered?

WasabiFan, 2020-03-14:
I think we agree this is a bug, so in some sense "yes", but I'd still consider this an open bug.

salkinium, 2020-03-14:
Yeah, sorry, I meant to say "Did you find a solution for this issue?"…

WasabiFan, 2020-03-14:
:laughing: yeah, I was using this file as a reference so once I realized why the initialize method wasn't being called I made it public and called it from my system initialization function. So my situation is solved now. Thanks!

closed 2021-09-29 by salkinium