modm-io/modm #328: STM32L4 I2C NACK should be handled by I2C_EV

mikewolfram, 2020-02-21:
Current implementation is handling the NACK in the error interrupt, but in reality it is an event and should therefor be handled there:

[image]

I made a quick a dirty fix and copied the part from the I2C_ERR handler to I2C_EV handler right after checking for stop. Works now.
Since I'm not an I2C expert maybe you could tell me, if that's the right position and especially if that needs to be handled in the template file to change it for L4 only? With some advice I'll provide a fix.

salkinium, 2020-02-22:
Related to this https://github.com/modm-io/modm/pull/305. cc @nesos

salkinium, 2020-02-22:
The L4 template is the place to fix it, yes.

mikewolfram, 2020-02-23:
I had a look into #305 before. That one simply enables the NACK IRQ. On systems which use a global I2C interrupt handler this works fine. But if there are I2C_ERR and I2C_EV IRQ handlers this would simply not handle the NACK flag.

Looked into few reference manuals which I had on my disk. F3, F4, G4, L4 and F7 all tell, NACK is a I2C event. Still not clear to me is, which MCU have the global IRQ handler and which not. Beside the table in the first post some manuals use this graph:
[image]
Reading the table above could also mean, there are event and error interrupts as well as a global interrupt.

Anyway it looks to me the right place to handle NACK is the event IRQ handler. I'll prepare a PR and we can discuss whether the code is in the right place and does the right thing.

PR #331 (2020-02-28): Move the check for I2C NACKF to the event handler
As explained in the reference manual, NACK is handled by the I2C_EV handler and not the
I2C_ERR handler.
Fixes #328.
closed 2020-02-28 by salkinium