modm-io/modm #330: STM32L4 ADC returns wrong values when gcc optimization not -O0

mikewolfram, 2020-02-27:
Turning on optimization in gcc (e.g. -Og) results in a weird behavior of the ADC. The very first conversion is fine, all of the following return useless data. Example is reading the temperature sensor, right after start it reads about 34C, the next reading is about 800C.

The only difference I found is the ISR register of the ADC. After reset it is 0, after the conversion it is 0x07 (EOS, EOC and EOSMP). Starting the next conversion clears on EOSMP and OVR, EOC is implicitly cleared.

Modifying startConversion() method like this
```cpp
void
modm::platform::Adc1::startConversion(void)
{
	// TODO: maybe add more interrupt flags
	acknowledgeInterruptFlag(InterruptFlag::EndOfRegularConversion | 
			InterruptFlag::EndOfSampling | InterruptFlag::Overrun);
	// starts single conversion for the regular group
	ADC1->CR |= ADC_CR_ADSTART;
}
```
resolves the issue.

Question is now, if it is ok to simply add the flag to the stm32f3 ADC class for all types? Maybe somebody could check it? Otherwise I'd add a 'if' to the template file to change it for L4 only.

I have absolutely no idea, why without optimization everything is fine. Searching Google brings up few results, that the problem of the ADC with optimization turned on is known, but the solutions provided there didn't work for me. For instance the DIFSEL register is always 0, also a different tool chain didn't resolve the problem.

Using ST gcc 7-2018-q2-update.

rleh, 2020-02-27:
Do I understand your change correctly?
```diff
void
modm::platform::Adc1::startConversion(void)
{
	// TODO: maybe add more interrupt flags
+	acknowledgeInterruptFlag(InterruptFlag::EndOfSampling | InterruptFlag::Overrun | InterruptFlag::EndOfRegularConversion);
-	acknowledgeInterruptFlag(InterruptFlag::EndOfSampling | InterruptFlag::Overrun);
	// starts single conversion for the regular group
	ADC{{ id }}->CR |= ADC_CR_ADSTART;
}
```

---

ADC_ISR = 0x7 means ADRDY, EOSMP and EOC are set.
[image]

---

> a different tool chain didn't resolve the problem

Which toolchains have you tested?

mikewolfram, 2020-02-27:
Ah sorry, meant 0x0c. Looking into bits all day and got confused by the notation there:
[image]
Will change it in my text above.

salkinium, 2020-02-27:
Smells like a write-buffer issue, try adding a `__DSB(); __ISB(); ` as the last line in the interrupt.

salkinium, 2020-02-27:
Maybe `acknowledgeInterruptFlag()` needs to wait for the interrupts to be really cleared, by adding a data-synchronization barrier (`__DSB()`) before starting the conversion? The Cortex-M4 has a write buffer of size 1, which can take several cycles to write depending on bus utilization.

salkinium, 2020-02-27:
But there is already a TODO, so maybe that's just it?

mikewolfram, 2020-02-27:
I think clearing the EOS flag as well doesn't hurt. So the ISR register is in same state like when MCU started and it worked for the first time. Not sure if a __DSB() would help as a constant is written to the register and it shouldn't make a difference if we write a 0x12 or a 0x16.

salkinium, 2020-03-12:
Any progress on this issue?

mikewolfram, 2020-03-12:
Using it this way for two weeks without issues. Also had a look into the generated assembler and I think, the reason why it works in debug build is the amount of instructions between starting the conversion and checking if it's finished. With -Og there are only a few instructions, the code is already pretty good optimized. So I'd create a PR as explained above.

mikewolfram, 2020-03-12:
BTW, I use the two gcc provided in STM32CubeIDE (which is a 7.2) and also have a gcc 9.2 installed from MacPorts.

PR #347 (2020-03-13): Clear ADC interrupt flags when starting a conversion
Clears the interrupt flag EndOfRegularConversion as well when a conversion is started. Thix fixes issues with wrong values read from the ADC when compiling
with optimization turned on.
Fixes #330.
closed 2020-03-13 by salkinium