modm-io/modm #341: [question] What are the base class static member functions for?

dhebbeker, 2020-03-10:
I am recently working in particular with the modm timers. I noticed that they inherit form the `GeneralPurposeTimer` and / or the `BasicTimer` classes. The static member functions of those classes are not defined.

What are those declarations good for?

salkinium, 2020-03-10:
I assume these were just for having a clean API documentation interface and for inheriting common definitions from basic to general to advanced.
Maybe @ekiwi knows more?

ekiwi, 2020-03-11:
>  Maybe @ekiwi knows more?

We used to do this in `xpcc` in order to document the interface (like concepts). This was never enforced by the compiler afaik. You can probably get rid of this if you want to.

salkinium, 2020-03-12:
Ah yes, so this was like a platform specific HAL interface, with C++20 they could be described as proper Concepts and thus actually compiler-enforced. C++20 will bring a lot of good refactoring to modm, there's so much really great stuff in it!

Did this answer your question? (and do you want to close this issue?)

dhebbeker, 2020-03-12:
Thank you everybody for answering my question.

closed 2020-03-12 by dhebbeker