modm-io/modm #349: Documentation von :test module missing

rleh, 2020-03-14:
Code and modules in ['test/'](https://github.com/modm-io/modm/tree/develop/test) do not show up in the [module reference](https://modm.io/reference/documentation/) and [code/api codumentation](https://docs.modm.io/).

salkinium, 2020-03-14:
They also don't show up in lbuild discover, because they are technically part of a different repository `modm-test`. This was intentional: https://github.com/modm-io/modm/pull/110.
Additionally, these modules do not have any documentation to show 😬.

I would be quite happy to have test documentation (also of the `:unittest` module/API!), but they should probably be separate from the already existing documentation (particularly they should not be "hidden" somewhere deep in the module jungle).

salkinium, 2020-03-14:
Oh, actually [the `:unittest` API is available](https://docs.modm.io/develop/api/stm32f103zgt7/group__modm__unittest.html), but probably not a good sign, if I only noticed this now 🙄.

rleh, 2020-03-14:
> Oh, actually the :unittest API is available, ...

I know.

But I would like to see e.g. both (!) spi mocks ([`modm::platform::SpiMasterMock`](https://github.com/modm-io/modm/blob/develop/test/modm/platform/spi/mock/spi_master.hpp) and [`test::SpiDevice`](https://github.com/modm-io/modm/blob/develop/test/modm/driver/adc/spi_device.hpp)) in the API documentation.

What about adding the `test/repo.lb` to the [docs.modm.io](https://docs.modm.io/)-generator?

rleh, 2020-03-14:
> What about adding the test/repo.lb to the docs.modm.io-generator?

Hm, does not work out of the box:
```
Generating documentation for hosted-linux ...
Error generating documentation for device hosted-linux: Repository(modm) already has a submodule named 'modm:ros'!
Hint: This Module defined in '../../home/user/modm/ext/ros/module.lb':
    >>> Module(modm:ros)   # ROS Library
conflicts with this Module defined in '../../home/user/modm/ext/ros/module.lb':
    >>> Module(modm:ros)   ROS Library
```

salkinium, 2020-03-14:
Hm, bummer. Let me check, I'm also not sure if `:docs` generates the right docs for other repos either (because the question is *where* do you generate other repos docs? Centrally into modm?).

However, you definitely need to add all the missing `@ingroup` to the headers though!

salkinium, 2020-03-15:
I mean it work ok-ish if you simply add the test repo to the docs_modm_io_script:
```py
builder.load([Path(modm_path) / "repo.lb", Path(modm_path) / "test" / "repo.lb"])
```
That simply results in the right module structure, but of course they're all empty cos of the missing `@ingroup`s. Can you replicate that?

The module docs tree generator has other issues related to hardcoding one too many strings, but considering the `modm-test` module docs are empty anyways, not a priority for me right now.

rleh, 2020-03-16:
I modified the script the following way and came up with the error above:
```
builder.load(Path(modm_path) / "repo.lb")
builder.load(Path(modm_path) / "test" / "repo.lb")
```
I will fix it, test it, add missing `@ingroup`s and create a PR in the next days.

PR #354 (2020-03-17): Documentation improvements
- [x] Test mode with targets from real target list (e.g. stm32f103zgt7 instead of stm32f103c8t6)
- [x] Exclude unused architecture modules, see https://github.com/modm-io/modm/issues/353#issuecomment-599826924
- [x] Build documentation for test (include `test/repo.lb` lbuild repo)
- [x] Allow search crawlers on docs.modm.io

Fixes #349
closed 2020-03-17 by salkinium