modm-io/modm #353: remove declarations of undefined static member functions

dhebbeker, 2020-03-16:
> We used to do this in `xpcc` in order to document the interface (like concepts). This was never enforced by the compiler afaik. You can probably get rid of this if you want to.

_Originally posted by @ekiwi in https://github.com/modm-io/modm/issues/341#issuecomment-597427579_

I suggest to remove the declarations of static member functions which are never defined for the following reasons:

* remove redundant declarations and documentation which may eventually diverge
* remove the potential impression (for new users like me) that there may be a generic / basic implementation which may be called using the base class

rleh, 2020-03-17:
The idea of the classes defined in `architecture` is to define an interface which then has to be implemented for every platform. E.g. in java the classes in `architecture` would be `interface`s, but C++ until 17 has no such feature.
I believe the usage of [C++ concepts](https://en.cppreference.com/w/cpp/language/constraints) would fix all the issues. Be concept are a C++20 feature and we will have to [wait at least until GCC 10](https://en.cppreference.com/w/cpp/compiler_support#cpp20) is released for ARM targets to switch to C++20.

rleh, 2020-03-17:
> remove the potential impression (for new users like me) that there may be a generic / basic implementation which may be called using the base class

Good point!
I guess we should not include all the `:architecture:*` modules in the documentation generator. Then only those `architecture` modules for which there are actual implementations should be included, since these are included by the `platform` modules as dependencies.
I will check this...

rleh, 2020-03-17:
Seems to work: 8c09487e5

dhebbeker, 2020-03-20:
Thank you @rleh for taking care of the documentation. 

My ultimate concern is a about the function declarations in the source code. I will leave this issue open for discussion and decision on (and potentially implementation of) the suggested actions.

rleh, 2020-07-06:
I consider the problem solved by 8194b880b75aa859d1a97ed4f498ef8ecac2a337 (#354).

closed 2020-07-06 by rleh