modm-io/modm #359: Restructure examples

rleh, 2020-03-19:
### Problems / possibilities for improvement

* Boards modules contain lots of duplicate code.
* Only 33 dev boards are currently supported.
  * Little effort to support all 47 ST Nucleo boards.
  * Little effort to support Arduino derivates.
* One example only run one one single dev board without modifications.
* Examples are hard to use for on-target software testing.
* Many examples are hardly documented.

### Restructuring modm examples

* Unify board modules: Generate (e.g.) all nucleo board form a single lbuild module.
* Extend board.hpp: Add "default" Spi, I2c, Uart, Gpio, ... (*interfaces*)
  * Should (if possible) use same pins on same form-factor dev boards.
  * Availability and number of *interfaces* as lbuild module properties on which the examples are then dependent.
  * Document *interfaces* [similar to mbed](https://os.mbed.com/media/uploads/bcostm/nucleo_f303k8_2017_10_10.png)
* Generated (automated from CI) all examples into https://github.com/modm-io/modm-examples to keep the first steps with modm easy.
* Make examples usable for automated integration tests, e.g. with a `PASS`/`FAIL` in the serial log.

rleh, 2020-03-19:
### List of Nucleo Boards

* nucleo-f030r8
* nucleo-f031k6 (supported in modm)
* nucleo-f042k6 (supported in modm)
* nucleo-f070rb
* nucleo-f072rb
* nucleo-f091rc
* nucleo-f103rb (supported in modm)
* nucleo-f207zg
* nucleo-f302r8
* nucleo-f303k8 (supported in modm)
* nucleo-f303re
* nucleo-f303ze
* nucleo-f334r8
* nucleo-f401re (supported in modm)
* nucleo-f410rb
* nucleo-f411re (supported in modm)
* nucleo-f412zg
* nucleo-f413zh
* nucleo-f429zi (supported in modm)
* nucleo-f446re (supported in modm)
* nucleo-f446ze
* nucleo-f722ze
* nucleo-f746zg
* nucleo-f767zi
* nucleo-g070rb
* nucleo-g071rb (supported in modm)
* nucleo-g431kb
* nucleo-g431rb
* nucleo-g474re (supported in modm)
* nucleo-h743zi
* nucleo-h745zi
* nucleo-h753zi
* nucleo-h755zi
* nucleo-l010rb
* nucleo-l011k4
* nucleo-l031k6
* nucleo-l053r8
* nucleo-l073rz
* nucleo-l152re (supported in modm)
* nucleo-l412kb
* nucleo-l412rb
* nucleo-l432kc (supported in modm)
* nucleo-l433rc
* nucleo-l452re
* nucleo-l476rg (supported in modm)
* nucleo-l496zg
* nucleo-l4r5zi

salkinium, 2020-03-19:
In addition, we should have at least the Nucleo Pinouts available as machine readable data, so we can generate both BSPs and unittests for them [like this GPIO test](https://github.com/modm-io/modm/blob/develop/test/modm/platform/gpio/module.lb#L18).

salkinium, 2020-03-19:
> Generated (automated from CI) all examples into https://github.com/modm-io/modm-examples to keep the first steps with modm easy.

That's a great idea! Perhaps we can also generate https://github.com/modm-io/modm-template at the same time cos it would be basically have a mostly similiar structure?

rleh, 2020-03-19:
For the nucleo boards the only unavoidable difference is the `SystemClock` struct with all `static constexpr` frequencies and the `enable()` method.

Can we somehow get machine-readable data (from CubeMX?) to generate the code?

salkinium, 2020-03-19:
> Can we somehow get machine-readable data (from CubeMX?) to generate the code?

Yes, I found this: https://salkinium.com/stm32/clock/

But it's both matches too much and too little, because ST has some logic conditions inside their XML that I do evaluate, so there may be duplicates or missing nodes. It's pretty horrifying.

salkinium, 2020-03-19:
Ah, found the code: it's quite simple: https://github.com/salkinium/modm-devices/commit/5381b4bb71e49cedd19c2da06af32e10510bfc6d

closed 2021-09-29 by salkinium