modm-io/modm #360: External projects using modm

salkinium, 2020-03-19:
I want to build a list of projects that use modm, so that new people can perhaps explore "real" projects, not just the rather simple examples in modm.

Please add a link to your projects here, and perhaps some short feedback how well modm worked for you in this project. We're always looking for more feedback please!

rleh, 2020-03-19:
* https://github.com/roboterclubaachen/micro-motor
* Some other projects, but not public available...

mikewolfram, 2020-03-22:
I cannot provide any links since the code is hosted in-house. What we use it in our automation project as of now for:
* RFID reader based on NXP CLRC663 with CAN and optional 2.4" Touch TFT
* Conveyor controller
* Master controller acting as an interface between MES and automation hardware

The reason for us to select modm was, that it builds on modern C++, small footprint and still allows for lot of control. Had a look into mbed before and dislike, that there happens so much behind the scenes.

rleh, 2020-03-27:
Some more (now public accessible) projects:
* [Adapter from _Pixart PAT9125_ optical tracking sensor to AB-Style encoder signals](https://github.com/rx-viper/elektronik-und-software/tree/master/software/pixart-adapter)
* [Control and data acquisition for an experiment setup on a sounding rocket](https://github.com/rx-viper/elektronik-und-software/tree/master/software/onboard) (xpcc, not ported to modm)

strongly-typed, 2020-03-29:
Very good question! Gives a good opportunity to look retrospectively at what I did the last years. As with Mike, I sadly cannot share many of the developments as they were for clients or employers. In most cases, at least some code diffused into xpcc / modm.

I was doing some stuff myself and sometimes initiated other people to work with xpcc / modm. Here are some in arbitrary order:
* An exotic AVR32 project (now a fully deprecated platform for me)
* LED animation on the PhD hat of my partner.
xpcc was strong in providing good fades and gamma-correct colours
* WS18B20 LED installation for an exhibit for DLR Stuttgart
* Ultra Wideband Ranging library ([DW1000](https://github.com/roboterclubaachen/xpcc/pull/361) modules, done by @Maju-Ketchup)
* Interfacing an IMU to ROS (Claire during UROP at RWTH Aachen University)
* Interfacing Time of Flight sensors to ROS (Clemens at Robotise)
Driver was already there, the ROS serial lib was very handy and I could easily reuse from a former project
* ROS to CAN bridge using a cheap Nucleo board and external transceiver
Again, ROS serial was very handy to just plug together this in less than an hour.
* Interfacing four Maxon motor controllers to ROS (myself at Robotise)
  Reading out the hall sensors directly as encoders to get positional feedback (which the motor controller did not provide). @chris-durand took over this project and improved it a lot.
* Many PoC with sensors, microcontrollers and other stuff.

I probably need to extend the list later and watch out what I could do with some KUKA robots, a very large ABB 6-axis gantry robot at ERNE AG Holzbau here in Switzerland. I am probably looking into CANOpen and DeviceNet as I have to deal with some serious industrial stuff here.

mikewolfram, 2020-03-30:
> I probably need to extend the list later and watch out what I could do with some KUKA robots, a very large ABB 6-axis gantry robot at ERNE AG Holzbau here in Switzerland. I am probably looking into CANOpen and DeviceNet as I have to deal with some serious industrial stuff here.

I already started on CANopen and am also working on SECS protocol.

salkinium, 2020-04-27:
Ohh, I should probably add my own projects to this list 🤦‍♂️.

In reverse chronological order:

- [ELVA base board: CAN-to-{Stellwerk, Weiche, Achszähler, Signal}](http://www.via.rwth-aachen.de/elva.php): public soon (modm)
- [ELVA semaphores](http://salkinium.com/elva/signal): public soon (modm)
- [Navigation helmet: BNO055 IMU, NEO-6 GPS, NRF24 radio, WS2812B LEDs, NUCLEO-L432KC](https://github.com/salkinium/navimet) (modm)
- [Semi-accurate PID heater for my thesis](https://github.com/salkinium/hotbox) (xpcc)
- [iWall DMA multiplexor for 20k LEDs](https://hci.rwth-aachen.de/iwall) (xpcc)
- [Calibration robot for Corona](https://github.com/salkinium/calirona) (xpcc)
- [30pin connector short-circuit tester](https://github.com/salkinium/checkirona) (xpcc)
- [HSV color mixer](https://github.com/salkinium/lumicolor) (xpcc)
- [Colorful bicylce helmet (destroyed in electrical fire)](https://github.com/salkinium/ledfader) (xpcc)
- [A 8x8 flexible LED display on textiles](https://github.com/salkinium/textile_display) (xpcc)

Some unfinished proof-of-concept projects:
- [Modular STM32 Clock Tree in C++](https://github.com/salkinium/save-the-clocktrees)
- [Embedded graphics library](https://github.com/salkinium/upainter)
- all of modm, lol

WasabiFan, 2020-09-13:
Hey everyone! Myself and the rest of @uw-advanced-robotics has our own project to share :smile: 

At their core, our robots are all operated by STM32F4s; after a few years of working with... err... _less pleasant_ tools (_cough cough, Keil MDK, cough cough_) we adopted modm around a year ago. We've since rebuilt almost all functionality in the modm world, with the exception of a few robots' special features which drew the short straw due to this past year's truncated season.

I'd like to point out that our project is very much _not_ vanilla as far as modm is concerned; we have some constraints regarding approach-ability for people brand new to C++, as well as miscellaneous aspects of the dev workflow which involved some customization. In particular, pretty much all modm classes are wrapped in our own types -- this is primarily for test-ability. Our `SConstruct` is also a) almost entirely custom and b) written by people (read: me) who neither understand nor enjoy the tool in question. You have been warned :laughing: What I'm trying to say is: what works for us may not work well for others, and may indeed be suboptimal.

All of that being said, we hope others can benefit from our work!

Our main repo lives at https://gitlab.com/aruw/controls/aruw-mcb. We have a GitHub mirror at https://github.com/uw-advanced-robotics/aruw-mcb.

We've only recently made the repos public, so there may be some rough edges as we make other repos and their docs public. Making the issue tracker public is in our short-term plans. From here on out, we're developing in the open.

cc @MatthewMArnold

closed 2021-09-29 by salkinium