modm-io/modm #378: Are there plans to support stm32h7 devices?

EvilEli, 2020-04-15:
Are there any plans to support stm32h7 devices in the near future?

Playing around with lbuild and adapting the platform modules of `rcc` and `gpio` I (almost) managed to have a succesful build of the blinky example. The linking stage is still failing due to some memory section overflow. I guess i'll have to adapt something there as well. Probably you can point me somewhere in case I missed something obvious.

However, as far as I can judge, the h7 devices stand out a bit compared to the other stm32 devices.
At least the clock tree configuration seems to be quite different.
Would be interesting to know your plans regarding the stm32h7 devices.

salkinium, 2020-04-15:
Yes, I even have a STM32H7 lying around somewhere, but I got it *before* discovering the clock tree and power peripheral complexity and then was a bit demotivated to actually continue porting this. 

If you open a PR with your code, I'll *very happily* check out the linker issue (should theoretically use [this linkerscript template](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/stm32/linkerscript/stm32_idtcm.ld.in) due to caches etc.)
You can ignore the full clock tree for now and just use the 8MHz it boots with or hardcode some specific clock config in the SystemClock to test all the other peripherals. That way it isn't such a blocker right away.

salkinium, 2020-04-15:
Also note that the [STM32H7 data in modm-devices may not be accurate](https://github.com/modm-io/modm-devices/blob/develop/devices/stm32/stm32h7-43_53.xml#L30-L40), since the [memory layout is just… hardcoded](https://github.com/modm-io/modm-devices/blob/develop/tools/generator/dfg/stm32/stm_device_tree.py#L108-L119)… 🙈

EvilEli, 2020-04-16:
Ok I guess that could explain the linking issues :grimacing:

salkinium, 2021-09-29:
Single-core STM32H7 devices are now supported since https://github.com/modm-io/modm/pull/652.

closed 2021-09-29 by salkinium