modm-io/modm #380: Missing pyelftools in Linux installation guide

CaptainHose, 2020-04-23:
In order to get scons running (and therefore the examples) on Ubuntu 18.04 I had to also run 
`sudo apt-get install python3-pyelftools `
in addition to the steps at https://modm.io/guide/installation/
Maybe update tutorial?

salkinium, 2020-04-23:
pyelftools is a dependency of the `modm` tool, so doing `pip3 install modm` should install it.

This is at its root a Python2 vs Python3 issue, where lbuild and the modm tools and SCons uses Python3, but we don't explicitly support Python2 anymore. Unfortunately SCons does not explicitly allow to set the Python version, so it may default to using Python2 and therefore the pyelftools is missing for that.

You can force the python version though:
```
alias scons="/usr/bin/env python3 $(which scons)"
```

salkinium, 2020-04-23:
I'll update the docs to make this Python version issue more clear.

salkinium, 2020-05-04:
Linux installation guide also needs to make it clear that OpenOCD needs to be installed as the HEAD version!!! Due to the incredibly slow release cycle of OpenOCD the "current" release is lacking support for a lot of new targets.

PR #427 (2020-07-05): [docs] Update and fix small inconsistencies
Closes #380, closes #419.
closed 2020-07-05 by salkinium