modm-io/modm #385: examples_compile.py linux fails on macOS

strongly-typed, 2020-04-28:
Running `(cd examples && ../tools/scripts/examples_compile.py linux)` on macOS leads to the error that `objdump-9` is not found.

```
==========================================================================================
Building: linux/assert/main.cpp with CMake
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/sascha/Dev/modm/modm-str-darwin_ci/build/linux/assert/cmake-build-debug
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/sascha/Dev/modm/modm-str-darwin_ci/build/linux/assert/cmake-build-release
Scanning dependencies of target assert
[  9%] Building CXX object CMakeFiles/assert.dir/modm/src/modm/debug/logger/hosted/default_style.cpp.obj
[ 18%] Building CXX object CMakeFiles/assert.dir/modm/src/modm/io/iostream.cpp.obj
[ 27%] Building CXX object CMakeFiles/assert.dir/modm/src/modm/io/iostream_printf.cpp.obj
[ 36%] Building CXX object CMakeFiles/assert.dir/modm/src/modm/platform/core/assert.cpp.obj
[ 45%] Building CXX object CMakeFiles/assert.dir/modm/src/modm/platform/core/memory.cpp.obj
[ 54%] Building CXX object CMakeFiles/assert.dir/main.cpp.obj
[ 63%] Linking CXX executable assert.elf
[100%] Built target assert/bin/sh: objdump-9: command not found
make[3]: *** [CMakeFiles/assert.lss] Error 127
make[2]: *** [CMakeFiles/assert.lss.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [build] Error 2
```

I have `gcc` installed with `homebrew`:

```
brew info gcc
gcc: stable 9.3.0 (bottled), HEAD
```

The `xpcc` example also not compiles, but this due to the known and yet unresolved error with `zmqpp`.

Any idea how to fix this?

strongly-typed, 2020-04-28:
Arghh, that's related to #261. 

At least, the CMake example compiles with these fixes. zmqpp (and xpcc) is still broken

strongly-typed, 2020-04-28:
zmqpp is only part of the problem. The `linux/assert` example fails. The fixes from #261 are not yet merged ...

strongly-typed, 2020-04-28:
Concerning `zmqpp`: I can compile their example against their code. But it does not link against the brewed library with `gcc-9`.

```
git clone https://github.com/zeromq/zmqpp.git
cd zmqpp
mkdir build && cd build
➜  build git:(develop) brew info zeromq
zeromq: stable 4.3.2 (bottled), HEAD
➜  build git:(develop) brew info libsodium
libsodium: stable 1.0.18 (bottled), HEAD
➜  build git:(develop) g++-9 -Wall -I../src/ -I /usr/local/Cellar/zeromq/4.3.2/include -std=c++17 -L /usr/local/Cellar/zeromq/4.3.2/lib -L /usr/local/Cellar/libsodium/1.0.18_1/lib -g -o simple_client ../examples/simple_client.cpp ../src/zmqpp/*.cpp -lpthread -lzmq -lsodium
```

As `zmqpp` is not yet maintained properly anymore (they broke their CI, no one cares about releases), I would opt for removing `zeromq` from `modm` altogether.

It was used some time ago at RCA for a remote CAN debugger. I don't know if RCA still relies on `zeromq` in any way @rleh.

salkinium, 2020-04-28:
>  I would opt for removing zeromq from modm altogether.

:+1: I don't think the RCA uses it either, and is moving away from XPCC towards ROS. And our simulator broke a long time ago.

strongly-typed, 2020-04-28:
Ok, then the destiny of `zeromq` is settled here ... What about the [xpcc example](https://github.com/modm-io/modm/blob/develop/examples/linux/xpcc/basic/main.cpp#L19) using `zeromq`? Should it then be removed as well?

salkinium, 2020-04-28:
> Should it then be removed as well?

Yes, and XPCC is dying too. For new designs, use UAVCAN, since it's much more maintained.
Not even sure the RCA still uses XPCC, but I use it in the ELVA and thus should remain in here a little longer.

strongly-typed, 2020-04-28:
OK, then I prepared #386. But still compile error for the `CMake` example.

rleh, 2020-04-28:
> Not even sure the RCA still uses XPCC.

XPCC is not used anymore at @roboterclubaachen
We will use UAVCAN in the future, @se-bi is working on an integration with modm.

salkinium, 2020-04-28:
> We will use UAVCAN in the future, @se-bi is working on an integration with modm.

Ohhhh, fantastic!

se-bi, 2020-04-28:
> > We will use UAVCAN in the future, @se-bi is working on an integration with modm.
> 
> Ohhhh, fantastic!

Yeah, let's see. I had a pause on that, and in the meantime they released v1...

However, there might be some legacy stuff at the @roboterclubaachen using XPCC.
I am not sure with the latest python remote control stuff, if and how far zmq(pp) is necessary for that.

Running the older bots, we nevertheless need to checkout older version, so I guess: go ahead :wink:

salkinium, 2020-04-28:
> zmqpp is only part of the problem. The linux/assert example fails. The fixes from #261 are not yet merged ...

Good point.

salkinium, 2020-05-01:
Fixed via #261.

closed 2020-05-01 by salkinium