modm-io/modm #387: Timer broken on hosted / macOS

strongly-typed, 2020-04-28:
Consider this example on hosted/macOS:

```
#include <modm/debug/logger.hpp>
#include <modm/processing/timer.hpp>

using namespace std::chrono_literals;

modm::ShortTimeout timeout;

int
main()
{
	int ctr(0);
	timeout.restart(500ms);

	while(true)
	{
		if (timeout.isExpired()) {
			MODM_LOG_DEBUG << "ping " << ctr++ << modm::endl;
			timeout.restart(500ms);
		}
	}
}
```

It spit out numbers without delays.

[Some code](https://github.com/modm-io/modm/blob/develop/src/modm/platform/can/canusb/canusb_impl.hpp#L104) that deals with real world peripherals relies on that.

Probably related to #374. I'll do a bisect soon.

strongly-typed, 2020-04-28:
Broke from commit f1021b0 to c63a5366e. I cannot compile the commits in between.

salkinium, 2020-04-28:
Maybe the cast from int64_t to uint32_t is not the best idea… I didn't even test this \*shame\*.

salkinium, 2020-04-28:
(can reproduce)

salkinium, 2020-04-28:
hm, it works with a `modm::Timeout` instead of a ShortTimeout, so I suspect issues with casting from uint32_t to uint16_t ms.

salkinium, 2020-04-28:
Yep, found the bug. I need to add some tests for this though…

PR #389 (2020-04-28): [timer] Fix time_point cast for short versions
Whoops, implicit casting during comparisons is quite horrible.

Fixes #387.
closed 2020-04-28 by salkinium