modm-io/modm #397: "ERROR: In Module 'modm:platform:core':" on lbuild build

mfp20, 2020-05-06:
If I enable modm:architecture:i2c, I get the following error

```
ERROR: In Module 'modm:platform:core': 
In file '../../modm/src/modm/platform/core/avr/module.lb:65' in function 'build':
        env.template("delay_impl.hpp.in")

Error in template '../../modm/src/modm/platform/core/avr/delay_impl.hpp.in:47':
	}) : ({
%% if f_cpu <= 6e6
		constexpr int32_t cycles = (F_CPU / 1e6);
		for (int16_t _us = us; _us > 0; _us -= cycles)

TemplateSyntaxError: expected token 'end of statement block', got 'e6'
```
project.xml

```
<library>
	<repositories>
		<repository><path>../../modm/repo.lb</path></repository>
	</repositories>
	<extends>../common.xml</extends>
	<options>
		<option name="modm:target">atmega328p-pu</option>
                <option name="modm:platform:core:f_cpu">20000000</option>
		<option name="modm:build:build.path">../../../build/mcu/atmega328p/</option>
		<option name="modm:build:scons:include_sconstruct">False</option>
	</options>
	<modules>
		<module>modm:platform:clock</module>
		<module>modm:platform:uart:0</module>
	</modules>
</library>
```

and its ancestor common.xml

```
<library>
	<options>
		<option name="modm:build:project.name">projectname</option>
	</options>
	<modules>
    		<module>modm:platform:core</module>
    		<module>modm:architecture:gpio</module>
    		<module>modm:architecture:adc</module>
    		<module>modm:architecture:1-wire</module>
    		<module>modm:architecture:uart</module>
    		<module>modm:architecture:i2c</module>
    		<module>modm:architecture:i2c.device</module>
    		<module>modm:architecture:spi</module>
    		<module>modm:architecture:spi.device</module>
		<!-- inter-board communication -->
		<module>modm:communication:ros</module>
		<!-- generic software interfaces, slow bit banging -->
		<!--<module>modm:platform:1-wire.bitbang</module>
		<module>modm:platform:i2c.bitbang</module>
		<module>modm:platform:spi.bitbang</module>-->
		<!-- multithreading -->
    		<module>modm:processing:protothread</module>
		<!-- debug, build system and docs -->
		<module>modm:debug</module>
		<module>modm:build:scons</module>
    		<module>modm:docs</module>
	</modules>
</library>
```

salkinium, 2020-05-06:
Hm, i guess `6e6` is not a valid number in Jinja2. You can change it to `6000000` and it should work. (and open a PR to fix it)

salkinium, 2020-05-06:
I wonder why the CI didn't catch this, it's definitely generating and compiling this module… 🤔

mfp20, 2020-05-06:
I tried to replicate the issue restarting from git clone, and it comes up on the fresh install as well. Maybe my python env? (I used the same env)

salkinium, 2020-05-06:
Dunno, I don't think there's any other use of `e` in the templates so just replacing it with the number is easier than to figure out environment issues.

mfp20, 2020-05-06:
And yes, replacing 6000000 in modm/src/modm/platform/core/avr/delay_impl.hpp.in fix the issue.

Do I have to open a PR? Where?

salkinium, 2020-05-06:
You can even do it online: https://github.com/modm-io/modm/edit/develop/src/modm/platform/core/avr/delay_impl.hpp.in then click on the edit icon upper right.

mfp20, 2020-05-07:
ahhhhhh... PR = Pull Request!!!! So you want for me to fork, correct the bug, and pull request!

closed 2020-05-07 by salkinium