modm-io/modm #406: SPI data size written to wrong register on L431

mikewolfram, 2020-05-19:
Hi,

figured out that the function to set the SPI data size on a L431 is wrong. The bits are written to CR1, but must be written to CR2. SpiXHal::initialize() does it right, but SpiXHal::setDataSize() does it wrong.

Could it be that the check for data size in features needs to be negated?
```c++
void inline
modm::platform::SpiHal{{ id }}::setDataSize(DataSize dataSize)
{
%% if "data-size" in features
	// TODO: implement as set/reset bit
	SPI{{ id }}->CR1 = (SPI{{ id }}->CR1 & ~static_cast<uint32_t>(DataSize::All))
										 | static_cast<uint32_t>(dataSize);
%% else
	SPI{{ id }}->CR2 = (SPI{{ id }}->CR2 & ~static_cast<uint32_t>(DataSize::All))
										 | static_cast<uint32_t>(dataSize);
%% endif
}
```
When you look in initialize() then the check is
```
%% if "data-size" not into features
```

Thanks.

salkinium, 2020-05-19:
It could also be that modm-devices has it wrong. The features are manually added, which in practice introduces copy-paste errors. Or the L4 hardware has a different layout altogether.

mikewolfram, 2020-05-19:
Already checked the device XML file, which has the feature flag set.

Like I said, in initialize() it checks for "not data-size in features" and writes the data size into CR2, in setDataSize() it is opposite. I'm not so familiar with other STM32 and right now I have no time to check a few more data sheets/reference manuals, but I'd expect the issue to be resolved when changing the %%if in setDataSize().

I guess nobody has been affected by this yet. The data size is set during initialization and nobody would change that later on. I have a problem with the ILI9341 TFT controller, which is rather strange. Writing data to it has no issues at all, but when reading data from it, then it wants a dummy clock cycle between sending the register to read and returning the data. So I decided to change the data size to 9 bits, shifting the value to send to the left by one bit and before reading the answer I set the data size back to 8 bits.

salkinium, 2020-05-19:
Ah ok, the initialize method does indeed write in CR1. It's very likely my mistake when I added features and refactored this driver but didn't test the data size. Wanna open a PR?

mikewolfram, 2020-05-19:
> Wanna open a PR?

Will do.

PR #407 (2020-05-23): [spi] Fix writing SPI data size into correct register
Also adds the missing method to the SPI master class.
Fixes #406.
closed 2020-05-23 by salkinium