modm-io/modm #409: ADC initialize() methods

rleh, 2020-05-23:
There are currently four different ADC platform drivers for STM32 in modm.
I would like them to use the same interface to make porting easier.

```cpp
// STM32F103, STM32F373, STM32F429, STM32F746
template<class SystemClock, frequency_t frequency = MHz(10), percent_t tolerance = pct(10)>
static void
initialize();

// STM32F072, STM32G071
template<class SystemClock, ClockMode mode, frequency_t frequency = MHz(1), percent_t tolerance = pct(10)>
static void
initialize();

// STM32F303
static void
initialize(const ClockMode clk=ClockMode::DoNotChange, const Prescaler  pre=Prescaler::Disabled, const CalibrationMode  cal=CalibrationMode::DoNotCalibrate, const bool blocking=true);

// STM32L433, STM32G474
static void
initialize(const ClockMode clk=ClockMode::DoNotChange, const ClockSource  clk_src=ClockSource::SystemClock, const Prescaler  pre=Prescaler::Disabled, const CalibrationMode  cal=CalibrationMode::DoNotCalibrate, const bool blocking=true);
```

What do you think about moving the clock mux configuration (which is usually shared between several instances) into the RCC or SystemClock code?

I would also suggest to remove the calibration from `initialize()` and let the user do it manually.

salkinium, 2020-05-24:
Agreed, the initialize methods are very confusing especially with the shared clock(s) among the ADCs. But perhaps a `AdcCommon{12|34}` class would make more sense? There are more features than just the clock shared among the ADC.

Thinking a little further, the problem of choosing the right ADC (instance, channel) for a pin reminds me of the DMA resource problem, where the specific optimal ADC (instance, channel) choice depends on the whole system configuration. In addition there is also the whole (semi-)automatic chained(?) conversion thing going on here, with a lot of scheduling options (interlacing, injection etc) plus DMA, which is in its complexity definitely beyond a "simple" C++ API. So the `AdcCommon` and `Adc{instance}` classes should just provide a basic HAL abstraction to simplify complex ADC configurations, right?

rleh, 2020-05-24:
I see the ADC api im modm currently being useful for most simple use-cases. Read one (or multiple) ADC channels using polling or interrupts.
Complex scenarios, e.g. master/slave mode and/or triggering from other peripherals, will always require a deep dive into the ST reference manual (and then writing to registers directly is ok for me).

I like the Idea of adding a `AdcCommon12`/`345` class 👍 
This class then configures the ADC common registers and RCC registers, not?

```cpp
AdcCommon345::initialize<SystemClock, 1_MHz, ...>();
Adc4::initialize<...>(...);
Adc4::read(channel4, ...);
```

salkinium, 2020-05-24:
> Complex scenarios [...] will always require a deep dive into the ST reference manual

Agreed.

> This class then configures the ADC common registers and RCC registers, not?

Yes, exactly! And also deals with the shared interrupts that are currently also wrongly part of the per-instance API.

closed 2021-09-29 by salkinium