modm-io/modm #411: Support for custom linker script

mcbridejc, 2020-06-02:
Hi, 

Is there any support for -- or intention for -- proving a custom linker script? I'd like to be able to carve out a couple sectors of flash for non-volatile app storage. For example: put the ISR vectors in the first sector, then use two of the small 16K sectors at the top of flash for app config, and then place the application code in the remaining flash sectors. 

I'm not seeing any way to disable the `-Tlinkerscript.ld` option in modm, which is all I need really to provide my own in my CMakeLists.txt. I guess I will simply disable the generation of this flag for now; would there be interest in a PR to make this an option that can be set in project.xml?

salkinium, 2020-06-02:
Yep, that's something I would like to have a generic solution for, it's just difficult to do.
You can *add* sections via the [`:platform:cortex-m:linkerscript.*` collectors](https://modm.io/reference/module/modm-platform-cortex-m/#collectors) (used to [add external memory](https://github.com/modm-io/modm/blob/develop/src/modm/board/disco_f469ni/module.lb#L47-L91)), but not change or move stuff around. 

modm also requires a lot of the tables that collect distributed information from your project, so you cannot just use a vanilla linkerscript. The only way is to *derive* your linkerscript from the generated one and then tell the build system to use that.

Maybe we should add a PathOption to the `:build` module, that is set to `modm/link/linkerscript.ld` by default, but lets the user overwrite with their own path? That way you could completely replace the linkerscript but just "extend" the modm linkerscript via the [`INSERT [AFTER|BEFORE]` mechanism](https://sourceware.org/binutils/docs/ld/Miscellaneous-Commands.html)?

Do you want to hack this together? The linkflags are set from here ([avr](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/avr/module.lb#L38)) ([cortex-m](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/module.lb#L319)), and the `{linkdir}` substitution is resolved here ([cmake](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/cmake/module.lb#L48)) ([scons](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/scons/module.lb#L69)) in the build modules. You would change this to `-T{linkfile}` and substitute them via the PathOption.

salkinium, 2020-06-02:
> that is set to `modm/link/linkerscript.ld` by default

Probably better to set it to an empty path as default, to be able to distinguish between the default and the user path (the path needs to be *relative* to the build script, not hardcoded absolute).

salkinium, 2020-06-02:
Would probably be more consistent to put the option in the `:platform:cortex-m` module, because that's where the linkerscript collectors are already. avr-gcc provides their own hardcoded linkerscript anyways so there's no replacing that, and the Hosted targets don't have custom linkerscripts either.

mcbridejc, 2020-06-04:
Thanks for the pointers, that helps a lot since I haven't gotten too familiar with lbuild yet. I will try to put something together for this so that I can copy/modify the auto-generated linker script and use my version instead of the auto-generated one -- while still generating the default on `build`.

salkinium, 2020-06-08:
Are you making progress? If you're stuck just ping me.

mcbridejc, 2020-06-10:
Sorry I switched tasks and didn't get back to it until today. The PR above is working for me using CMake, and it built successfully (seems to work) with a test scons project both with and without a linkerscript.