modm-io/modm #412: cmake issues

jasa, 2020-06-05:
I had some issues with file globbing and other things in the generated cmake files so i ended up rewrite the hole thing based on [this project](https://github.com/lefticus/cpp_starter_projectl).
It's a more modern use of cmake and it was really worth the effort. It is much simpler to include external modules and it gives tremendously faster compile times (recompiles in milliseconds then cached) and it has multi core support.

**It includes the following features:**
- ccache
- link time optimization
- precompiled headers
- doxygen

**Static checks:**
- clang-tidy
- cppcheck
- include what you use

is there any interest in implementing it in lbuild?

salkinium, 2020-06-05:
> is there any interest in implementing it in lbuild?

YES! I know the CMake generator isn't the best way to do it, but I just don't know better. There's so much conflicting information on the interwebs regarding CMake best practice and I lack the experience to judge it.

I'd be very, very happy if you took a look at it, I will of course help you with the lbuild stuff! I'm particularly interested to see an easier way to include external modules, as this problem has come up many times before.

jasa, 2020-06-06:
Hi Niklas

Here is a initial cmake attempt on github you can try. I have only tested
it on Linux and i will test it on a virtual Windows later. There are still
some issues to solve but it should work fine for now.
https://github.com/jasa/modm_cmake

These 4 configurations is available default: Debug, Release, RelWithDebInfo
and RelSizeMin. In ccmake the enter key can toggle between then

A basic example is:

mkdir build
cmake -DCMAKE_BUILD_TYPE=MinSizeRel ..          ( or ccmake.. )
make -j4

[image: Screenshot from 2020-06-06 16-40-49.png]
<https://github.com/jasa/modm_cmake>

salkinium, 2020-06-06:
Thanks, I'll give it a try in the next few days, currently fairly busy with work/uni.

Does LTO work for you? I was [under the impression that custom linkerscript support was lacking](https://twitter.com/salkinium/status/959488890133450752), but that was >2 years ago…

jasa, 2020-06-06:
Hi Niklas

Yes, normally i don't have any problems with LTO until now. I noticed scons
(indirectly)  fails with LTO, some archive operation needs extra parameters.

Your build script produces a .lss file with objdump which i don't know the
use of. It fails then LTO is enabled: *arm-none-eabi-objdump: DWARF error:
could not find abbrev number 108*

add_custom_command(TARGET ${CMAKE_PROJECT_NAME}
    POST_BUILD
    COMMAND ${CMAKE_OBJDUMP}  -x -s -S -l -w ${CMAKE_PROJECT_NAME}.elf >
${CMAKE_PROJECT_NAME}.lss)

LTO also reveals that _exit is define several times. I remember another
project where i wrote my own libc stub and didn't rely on nosys.specs for
the same reason.

salkinium, 2020-06-06:
The listings file is completely optional, it's just there to create a disassembly so you can check how the compiler translates your code. It's very helpful for understanding C/C++ overheads etc.
The `.bin` and `.hex` commands are also optional, they are only shortcuts for creating bootloader compatible files. I wasn't sure how to properly expose this optionality in CMake.

I'd very much like LTO enabled in modm, I'll have a closer look next week. Ideally the toolchain is fixed and it just works™, then we can just set the flags as default.

jasa, 2020-06-06:
.lss file are disabled then LTO are enabled
_exit conflict solved. ext/gcc/cabi.c is replaced with ext/gcc/cabi.cpp and
nosys.specs are dropped.

And then i found this:* Bug 88643*
<https://gcc.gnu.org/bugzilla/show_bug.cgi?id=88643> - -Wl,--wrap not
supported with LTO
so - -Wl,--wrap is not guaranteed to work with LTO for now.

salkinium, 2020-06-06:
> so `--Wl,--wrap` is not guaranteed to work with LTO for now.

That's a shame, but we only use it to wrap newlib malloc with our own implementation in `:platform:heap`, which is optional. LTO will be most useful on tiny devices like STM32L011, where you don't want to have a heap anyways, so I think that limitation is fine, so we can simply enable LTO if `:platform:heap` is not selected.

> ext/gcc/cabi.c is replaced with ext/gcc/cabi.cpp and nosys.specs are dropped

`modm_assert` works in C too, you can therefore keep it `cabi.c` instead of going to C++ land. You should define all functions as `modm_weak`, so that the application may still overwrite them.

`_sbrk` is also weakly defined in `modm/platform/core/no_heap.c`, and assigned a [specific section in the linker script to trigger a link error](https://modm.io/reference/module/modm-platform-heap-stm32/#heap-is-not-implemented-error). This is to prevent accidental use of the heap. Your nosys replacement therefore should not define `_sbrk`.

jasa, 2020-06-07:
Now I really nailed it (i thing). All target specific features are removed
from CMakeLists so they can be included in test scripts etc. I think
beginners will love it, all the complicated stuff has disappeared.

Your right about cabi, normally i would use ISO C++ style [[weak]] but now
it is in modm style. I really hate hard tabs, the next thing will be fixing
.clang-format  so it can handle the dirty work. :-)

Has modm any coding guide

salkinium, 2020-06-07:
> I really hate hard tabs

Yeah, me too. We briefly considered changing them, but we would need to touch every file in modm, and that would mess with git blame…

> Has modm any coding guide

Yes, but it's not written in stone and clang-format cannot do this exact style, so we'll have to adapt it to what clang-format can do.
https://github.com/modm-io/modm/blob/develop/docs/coding_convention.md

jasa, 2020-06-07:
Sorry I mailed a half mail by accident earlier.

I will leave the main branch stable for now so you can test. The flags are
correctly updated now then LTO state is changed (you need to update for the
latest changes)
there is no reason to implement the details before the concept is in place,
i will await your response.

List over cmake implementation.
* or ** = should be dynamic generated by lbuild
** = minor changes needed for AVR toolchain

 cmake/CompilerWarnings.cmake
 cmake/Doxygen.cmake
 cmake/StandardProjectSettings.cmake
 cmake/StaticAnalyzers.cmake
*cmake/TargetSettings.cmake
**cmake/ToolChain.cmake

*modm/CMakeLists.txt
*src/CMakeLists.txt

 and maybe a test template for users (i suggest catch2)
 test/CMakeList.txt

Notes:
I have made some changes in the smart_pointer.hpp/cpp due to some ugly
-Wcast-align warnings, you can use it or leave it.
a non reent version of _sbrk was needed.

~

salkinium, 2020-06-08:
I tried your repo, but there are issues with cross-compiling.
```
arm-none-eabi-gcc: error: unrecognized command line option '-mmacosx-version-min=10.14'
```

I remember this issue, it has something to do with need to set `CMAKE_SYSTEM_NAME` to Generic *before* calling project() or something.

```
include(cmake/StandardProjectSettings.cmake)
include(cmake/ToolChain.cmake)
include(cmake/TargetSettings.cmake)

# Set the project name to your project name, my project isn't very descriptive
project(myproject CXX C)
```

But that doesn't work either. It tries to use my system clang instead of arm-none-eabi-gcc.

```
clang: warning: optimization flag '-finline-limit=10000' is not supported [-Wignored-optimization-argument]
```

Regardless of these issues, I think your CMake files are much cleaner and more modular that what we have right now, so I would like to integrate your changes into modm.

Do you want to port your changes to lbuild and open a PR?

jasa, 2020-06-09:
A quick search brought me to this issue on stackoverflow
<https://stackoverflow.com/questions/52879026/cmake-cross-compile-on-macos-adds-macos-sdk-to-isysroot-in-flags-make>.
Can you try this CMAKE_OSX_SYSROOT on your system. I hope you can help me
out, I don't have a MAC available.

set(CMAKE_OSX_SYSROOT
/Volumes/xtool-build-env/armv8-rpi3-linux-gnueabihf/armv8-rpi3-linux-gnueabihf/sysroot)

I'm currently configuring a build environment on windows 10 with vscode for
testing, afterwards i will make a PR.

I found this in your cmake config but it doesn't make any sense to me. I
basically replaces a non existing string (on my platform) with nothing.
if(APPLE)
    # CMake adds system search paths on macOS for some reason
    if(CMAKE_C_LINK_FLAGS)
        string(REPLACE "-Wl,-search_paths_first" "" CMAKE_C_LINK_FLAGS
${CMAKE_C_LINK_FLAGS})
    endif()
    if (CMAKE_CXX_LINK_FLAGS)
        string(REPLACE "-Wl,-search_paths_first" "" CMAKE_CXX_LINK_FLAGS
${CMAKE_CXX_LINK_FLAGS})
    endif()
endif()

jasa, 2020-06-09:
It's now tested on Windows (vscode, MinGW) and Fedora 32 (vscode and
clion). No problems.

salkinium, 2020-06-09:
> I found this in your cmake config but it doesn't make any sense to me. I basically replaces a non existing string (on my platform) with nothing.

CMake on macOS  is wierd I guess… Google told me to put this there and then it just worked. 

> I hope you can help me out, I don't have a MAC available.

I'll try my best, however, our CI has a macOS, so you don't have to wait on me for the PR. Still lacking a Windows CI though.

jasa, 2020-06-10:
New status on Mac see: [https://github.com/jasa/modm_cmake](https://github.com/jasa/modm_cmake)

jasa, 2021-02-24:
This came to a halt in a previous project when I could use modm. It is almost implemented so why not finish it.

There are a few issues in the current design there isn't compatible with cmake and needs to be solved for a successful implementation. ideas, wishes and criticisms are welcome.

- Optimization flags (-Ox) is controlled by cmake and should be excluded, any idea howto handle them without breaking scons.
- Warning flags needs a separate group instead of being in ccflags.  I would like to make them private for modm so it's possible to define another set of warnings for your own code (easy to fix)
- Example projects don't have a src folder. It's not best practice (probably very bad) to have source file in the root folder with cmake and different ide's, It's not possible to have a CMakeLists.txt for the user source.
- External modules, complications???

rleh, 2021-02-26:
> Optimization flags (-Ox) is controlled by cmake and should be excluded, any idea howto handle them without breaking scons.

CMake has to behave the same way scons currently does:
* Size optimization `-Os`: `scons profile=release` for CMake `cmake -DCMAKE_BUILD_TYPE=Release`
* Debug optimization `-Og`: `scons profile=debug` for CMake `cmake -DCMAKE_BUILD_TYPE=Debug`

> Warning flags needs a separate group instead of being in ccflags.

That seems to make sense to me.

> Example projects don't have a src folder. It's not best practice (probably very bad) to have source file in the root folder with cmake and different ide's, It's not possible to have a CMakeLists.txt for the user source.

This is only the view of CMake. Other build systems do it differently.
With our example collection this makes no sense, so CMake has to adapt here somehow.

salkinium, 2021-03-02:
Regarding LTO: enabling LTO for embedded targets will break all our custom linkerscripts for either GCC or LLVM.
https://llvm.org/devmtg/2017-10/slides/LTOLinkerScriptsEdlerVonKoch.pdf

Unfortunately it's still not fixed:
http://llvm.1065342.n5.nabble.com/llvm-dev-LTO-with-Linker-Scripts-td142857.html

>> Has this work been upstreamed or implemented in any toolchains?
>
> AFAIK this effort stalled when Tobias changed jobs.

jasa, 2021-03-05:
> Regarding LTO: enabling LTO for embedded targets will break all our custom linkerscripts for either GCC or LLVM.

Not if LTO is disabled for these 2 file listed below and all other source file using HARDWARE TABLES like `src/modm/platform/gpio/enable.cpp`. I have done some tests with tlfs and it seems to works properly.

```
src/modm/platform/core/vectors.c
src/modm/platform/heap/heap_tlsf.cpp
```

salkinium, 2021-03-05:
Assertions also use linker tables, so does the heap tables and `.ramcode` and `.fastcode`. These are all fairly important for the API.

jasa, 2021-03-06:
My answer was maybe to bad. Tables initialized from linker script itself is not a problem like heap_table. The optimizer will remove the code from MODM_HARDWARE_INIT macro with LTO since it seems unused. To prevent this a simple attribute as [[gnu::used]] or a pragma can solve the problem. This is not a LTO problem, but has to be solved i the modm source.

The linker wrapper issue is bug where probably will be solved i the future. 

In new designs i cannot see why a thing like MODM_HARDWARE_INIT should be used, the same functionality can be achieved in pure C++.

salkinium, 2021-03-06:
> In new designs i cannot see why a thing like MODM_HARDWARE_INIT should be used, the same functionality can be achieved in pure C++.

Unfortunately it cannot, since the linker is the only place that can "collect" unknown objects from multiple compilation units. The same issue with `MODM_ASSERTION_HANDLER`, which also collects pointers to functions in a common linker section. It's not even a language feature, but rather a linker feature.

But it's irrelevant, we'll just test it with and without LTO and see what the differences in the binary/listing are. LTO would be really nice, but ultimately there are larger code size issues with modm, like using too many templates for everything resulting in code duplication…

closed 2021-05-08 by salkinium