modm-io/modm #421: Raw data for IOStreams

nesos, 2020-06-20:
I'm looking for a clean way to implement a HDLC Framer for use with modm. My first choice would be using IOStreams but so far it looks like only text is supported.
Adding a new "raw" mode would be relatively easy and would be useful for all kinds of stream processing, like framing, compression etc.
Would that be a desired thing to have in the IOStreams, or is there already a different way to output streams of raw binary data?

closed 2020-06-20 by nesos

salkinium, 2020-06-20:
Our IOStream is really more of an OStream, it may go away with C++20 `std::format` or a simpler `modm::format` implementation.

Not sure if IOstream is the right abstraction, you may want to create your own IO interface and prove a wrapper class for the supported peripherals.

salkinium, 2020-06-21:
Oh, just realized you closed this, I assume you found your answer already?

nesos, 2020-07-02:
Yes, I implemented my own stream interface to solve that issue. I realized I was heading in the wrong direction in the first place. Sorry for the inconvenience