modm-io/modm #424: Compiling and linking on ubuntu focal arm32v7 fails

ASMfreaK, 2020-06-22:
If one build an image from the following Dockerfile:
```Dockerfile
FROM arm32v7/ubuntu:focal

RUN \
    export DEBIAN_FRONTEND=noninteractive && \
    apt-get update && apt-get upgrade -y && \
    apt-get install -y \
        git openocd cmake make \
        libnewlib-arm-none-eabi gcc-arm-none-eabi binutils-arm-none-eabi \
        binutils-avr gdb-avr gcc-avr avr-libc gcc-avr avr-libc \
        libboost-all-dev libnet1-dev \
        python3-pip python3-pyelftools python3-lxml && \
    pip3 install lbuild scons

RUN \
    ln -s /usr/bin/python3 /usr/bin/python
```
One can the following error while building simple linux-hosted apps.

```
/usr/bin/ld: warning: release/modm/libmodm.a(iostream.o) uses 2-byte wchar_t 
yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may failed
```

If I can tell [this line](https://github.com/modm-io/modm/blob/ad56ca256c5eded997ff5bac84cabbf5ed3e4183/tools/build_script_generator/common.py#L217) is the the culprit - commenting it yields a clean build. I guess this is caused by linking to libraries built without `-fshort-wchar` that causes this issues.

Also weird stuff. On this compiler:
```cpp
static_assert(std::is_same_v<uint32_t, long unsigned int>); // fails
static_assert(std::is_same_v<uint64_t, long unsigned int>); // fails
// but
static_assert(sizeof(long unsigned int) == 4); // ok
static_assert(sizeof(long unsigned int) == 4); // ok
static_assert(sizeof(uint32_t) == 4); // ok
static_assert(std::is_same_v<uint32_t, unsigned int>); // ok
static_assert(std::is_same_v<uint32_t, long long unsigned int>); // ok
```
This make tests fail on `modm::Logger` to log `long unsigned int`, because there is no overload for it (only for uint32_t and uint64_t).

ASMfreaK, 2020-06-22:
Compiler version FYI:
```
$ g++ --version
g++ (Ubuntu 9.3.0-10ubuntu2) 9.3.0
```

ASMfreaK, 2020-06-22:
The latter problem is "fixed" by adding these lines to `iostream.hpp.in`, which compiles on both arm32v7 and x86:
```cpp
%% if options["with_long_long"]
#ifdef __arm__
#ifdef __linux__
	inline IOStream& operator << (const long unsigned int& v)
	{ writeIntegerMode(static_cast<uint32_t>(v)); return *this; }
#endif
#endif
```

salkinium, 2020-06-23:
Oh, interesting. I have the exact same issue on macOS, never occurred to me that this could be a compiler flag issue.

I would suggest only adding the compiler option on not-hosted targets, that would solve the problem, right?

PR #425 (2020-06-25): Patches for arm32v7
This fixes #424 

Changes in `src/modm/platform/core/cortex/assert.cpp.in` are only needed because of `-Wformat`:
```cpp
modm/src/modm/platform/core/assert.cpp: In function 'void modm_abandon(const modm::AssertionInfo&)':
modm/src/modm/platform/core/assert.cpp:58:71: error: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'uintptr_t' {aka 'unsigned int'} [-Werror=format=]
   58 |  if (info.context != uintptr_t(-1)) { MODM_LOG_ERROR.printf(" @ %p (%lu)", (void *)info.context, info.context); }
      |                                           [... cut]
closed 2020-06-25 by salkinium