modm-io/modm #429: Python error on Windows when running lbuild

mikewolfram, 2020-07-06:
Had to install a development environment on a Windows system today (typically use a MacBook) and came across an error in `modm/tools/build_script_generator/module.lb`. Running _lbuild_ on Windows results in an error, that WindowsPath isn't an iterable object. Line causing this is 36:

```
co = compile(tpath.read_bytes(), tpath, 'exec')
```
This showed up after including _modm:architecture:build_id_ module in project.xml.

To work around I simply return an empty string from this method right at the beginning, but I wonder, what would be the real fix?

Using Python 3.5 on a Windows 10 64bit.

rleh, 2020-07-06:
I can obviously not reproduce the error on my linux systems, but some ideas:
* [This stackoverflow answer](https://stackoverflow.com/a/36202611) suggests using `.name` on a Path object (I guess `tpath` is the problem):
  ```
  co = compile(tpath.read_bytes(), tpath.name, 'exec')
* Python 3.5 is pretty old, out of support since 2017. Could you try to reproduce the error with a more recent Python version?

We definitely need a Windows continuous integration tool, that checks at least some basic workflows.

mikewolfram, 2020-07-06:
Adding .name was my first idea too, but that raised another error.
Not sure if a newer Python will resolve that.
I’m a Mac user and now I’m faced with the fact to set up a development system on Windows. I’m neither a Windows nor Python expert.
Looks like nobody else is running Windows and including one of the tools.

mikewolfram, 2020-07-06:
What is the compile() function actually doing? Is it from Python or lbuild?

salkinium, 2020-07-06:
Oh uuhhh, this is a very dirty hack to extract the doc strings from the Python files in modm_tools.

This could/should be solved with regex instead though, it would probably be faster too. Lemme whip out my regex skills for this real quick.

salkinium, 2020-07-06:
Maybe `co = compile(tpath.read_bytes(), str(tpath), 'exec')` helps?

salkinium, 2020-07-06:
I have a replacement with regex that works and is faster.

salkinium, 2020-07-07:
Fixed in https://github.com/modm-io/modm/commit/58e9e464abeed0653576574d24935409936c1a0c.

closed 2020-07-07 by salkinium