modm-io/modm #447: Ethernet support / FreeRTOS+TCP

mikewolfram, 2020-08-23:
Ciao ragazzi,

have been working the last 2-3 days to get FreeRTOS+TCP working on a STM32F7. Created a modm::platform::Eth class as well as the functions required to initialize the interface and receive/send packets. Still pretty early, but for most cases it should be sufficient.

Question is now, how to get it into modm? Copy FreeRTOS+TCP and make it a module like FreeRTOS? Right now I have a modm::platform::Eth class and a separate modm::driver::Lan8720a class for the PHY.

Works/tested on a F765, so should be fine on all F7. Reading the FreeRTOS+TCP STM32 port (HAL based) there seem to be few differences on a F4. I don't have one to test, maybe if somebody wants to help on that?

Greetings from Bella Italia.
Mike

salkinium, 2020-08-24:
> Copy FreeRTOS+TCP and make it a module like FreeRTOS

Yes, but you'd copy the files via The `update.py` script into the ext repo: https://github.com/modm-ext/freertos-partial then add a new `:freertos:tcp` module or something like that.

salkinium, 2020-08-24:
You don't have to enable your port for everything, you can limit it to just what you tested.

mikewolfram, 2020-08-24:
Is there a reason that the update script is pulling a 2 years old clone of FreeRTOS?
```
if not Path("freertos_src").exists():
    subprocess.run("git clone https://github.com/cjlano/freertos.git freertos_src", shell=True)
```

salkinium, 2020-08-24:
Yes, it was the only official git clone available at the time. Should be https://github.com/freertos/freertos now.

mikewolfram, 2020-08-24:
Do you want me to update it in my upcoming PR or let it have its own?

salkinium, 2020-08-24:
You can update it.

mikewolfram, 2020-08-31:
Continuing on the ETH driver and the FreeRTOS TCP functions I ran into an issue. I have a Microchip I2C EEPROM with 48bit UID. That UID should be used as the MAC address. But I cannot read from I2C before the FreeRTOS scheduler has been started. A read from I2C is hanging forever, when executed in main() before scheduler start. Doing the read from inside a task it works without issues.

I have the impression that IRQs are not executed as expected, any hint what could be the reason?

Also tried to read the UID in a task and then call FreeRTOS_IPInit() from there, but that crashes the app. Maybe a stack issue?

salkinium, 2020-08-31:
Possibly an interrupt priority issue, where the I2C IRQ priority is too low? FreeRTOS could also perhaps disable interrupts between task creation and scheduler start?

salkinium, 2020-08-31:
There's a number of magic numbers that should probably be verified: https://github.com/modm-io/modm/blob/develop/ext/aws/FreeRTOSConfig.h.in#L123

mikewolfram, 2020-08-31:
> Also tried to read the UID in a task and then call FreeRTOS_IPInit() from there, but that crashes the app. Maybe a stack issue?

Indeed, had to increase the stack of that task to 1024 (256 words). Well, that solved one problem :-)

mikewolfram, 2020-08-31:
> FreeRTOS could also perhaps disable interrupts between task creation and scheduler start?

Found it. Yes, when creating a task the IRQ are disabled. The variable uxCriticalNesting is initialized with 0xaaaaaaaa, so when leaving a critical section the IRQ aren't enabled again. The scheduler will reset to 0 when starting a task.

salkinium, 2020-08-31:
Is this an initialization error? ie. Should FreeRTOS be initialized somehow *before* task construction?

There is currently [no way to initialize functions *after* memory/heap init, but *before* the init_array/static constructors](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/startup.c.in#L113-L129), so perhaps that may need to be added?

salkinium, 2020-08-31:
Although looking at the linkerscript by putting an init function into the section `modm_section(".init_array.A0000")` or similiar it should [be placed at the top of the init array](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/linker.macros#L207-L208).

salkinium, 2020-09-03:
Have your issues been solved? Is the integration working better now?

mikewolfram, 2020-09-03:
Still learning FreeRTOS, but making good progress. I don't know if FreeRTOS requires extra init/setup but it is really confusing when a modm::rtos::Thread is created before main and then nothing seems to be working in main, because xTaskCreate(), which is called in the constructor, disables all IRQ.

Anyway, working on simple HTTP server for our project, guess we could use it as an example later. Need to get things done until next week, so no time yet to prepare the ethernet files for inclusion in modm.

BTW: modm::rtos::Thread will create the task immediately and it is started right away when the scheduler starts. But I think we should have also Threads to be started later.

salkinium, 2020-09-03:
We honestly haven't used `modm::rtos` in a long time, so maybe we should completely rewrite it. Feel free to change what pains you, that's also why I'm interested in your work.

strongly-typed, 2020-09-14:
> modm::platform::Eth

Great to hear! Could you share the interface of that class? It should ultimately go to
[https://github.com/modm-io/modm/tree/develop/src/modm/architecture/interface](architecture/interface)
if it is possible to find an interface which works well with the integrated PHY and external PHYs like (outdated?) ENC28J60.

mikewolfram, 2020-09-14:
I didn't care too much about any external MAC (STM32F7 has internal MAC, but requires external PHY). When I'm back to the hotel tonight I'll prepare a draft PR with my code so we can start a discussion how to best integrate it. Maybe you could give some advise on that interface part?

I'm currently using the LAN8720A PHY (LAN8742 works too) and simple configuration, so it does always 10/100 auto negotiation. It uses DMA and FreeRTOS zero-copy buffers, MAC address is read from a Microchip 24AA02E48.

strongly-typed, 2020-09-14:
I meant MAC, not PHY. ENC28J60 is a MAC + PHY. Sorry, always get confused.

> I didn't care too much

That's totally fine for me, and having something to work on is better than nothing! I appreciate your efforts very much.

> Maybe you could give some advise on that interface part

I have no clue yet, working through the ENC28J60 right now. Maybe @salkinium want to have a say here? Looking forward to discussing any issues with you.

Are you using raw IP frames, your own higher level protocol or any sort of LWIP or IP stack? LWIP already defined some interface for different MACs. Let's see how this fits to modm.

salkinium, 2020-09-14:
> Maybe @salkinium want to have a say here?

Whatever works well with the network stacks. I don't see much sense in supporting Ethernet Frames without protocol (like UART), I would expect the main usecase to be with a stack.  (I don't have much of an opinion regarding internal/external MAC, due to lack of experience with Ethernet.)

The way that interfaces evolve in modm is usually tailored to a specific use-case in the beginning, and then as more use-cases get added, the interface is refactored. I think that's fine, we're not aiming for hard API stability.

strongly-typed, 2020-09-14:
> I don't see much sense in supporting Ethernet Frames without protocol (like UART)

I see ETH more like a faster CAN than an UART. The main difference to UART is that ETH has frames and some nice support features like arbitration, framing and checksums. Excuse the crude comparison, but I hope you got the idea.

> I would expect the main use case to be with a stack

Totally, but that shall not be limited to TCP/IP. I am thinking more of ETH as CAN as already mentioned. You can have hand-tailored CAN frames (think of the 1 Mbit motor CAN, so using Ethernet Frames with own protocol), you could make use of xpcc or even implement DeviceNet or CANopen on top of CAN frames.

> I think that's fine, we're not aiming for hard API stability.

Totally agree. First let's get something to work and then evolve from there.

mikewolfram, 2020-09-14:
> Are you using raw IP frames, your own higher level protocol or any sort of LWIP or IP stack? LWIP already defined some interface for different MACs. Let's see how this fits to modm.

Few weeks ago I added FreeRTOS-Plus-TCP to modm, which I use. Was testing LwIP first, but that's pretty old and FreeRTOS claims to have the faster/smaller implementation.

mikewolfram, 2020-09-14:
Opened #466. Wanted a draft PR, but created a real one. Sorry.

mikewolfram, 2020-09-15:
I ran into a problem, which is hard to reproduce/debug. Basically my app runs fine, but sometimes after a not determinable time the whole things crash because it is trying to free an invalid pointer. When trying to debug it, than the app runs for very long time without an issue. Then after a restart it might crash after few seconds of network traffic. Always the pointer to the ethernet frame inside the RX DMA descriptor of the MAC is invalid.

Is there a chance that somebody owns a F7 Nucleo board and we could make the example running there to see if that issue shows up there as well? I'm running on a custom hardware (the ethernet part is done 1:1 like on the Nucleo board).

I was thinking maybe the heap gets corrupted. I'm using the TLSF heap module. Would be great to get some support in investigation.

salkinium, 2020-09-15:
It could be an issue with thread-safety, since our heap wrappers aren't providing mutex support. (Someone™ should fix that… sorry!)

Apart from that you can try two things:

- You can change the [`:platform:heap:allocator` to `newlib`](https://modm.io/reference/module/modm-platform-heap-sam--stm32/#newlib), which is more tested than the TLSF one.
- You can change the [`modm_assert_continue_fail_debug`'s inside the TLSF wrapper](https://github.com/modm-io/modm/blob/develop/src/modm/platform/heap/cortex/heap_tlsf.cpp#L86) (and Newlib wrapper) to just `modm_assert`. That works in release mode too and is a much smaller change than compiling and running in debug mode.

salkinium, 2020-09-15:
Ah wait I remember locking wasn't part of the malloc wrapper, it was separate therefore not implemented in `:platform:heap`: I think you need these retarget locks?
https://gist.github.com/thomask77/65591d78070ace68885d3fd05cdebe3a

mikewolfram, 2020-09-15:
I added a modm_assert_continue_fail_debug to the tlsf malloc to make sure the pointer is always valid. Guess what? App was running for an hour without issue.
If there wasn't a SECS HSMS connection open, I'd silently restart every time it crashes. :-) Nobody would see it (except the SECS host).

mikewolfram, 2020-09-15:
> Ah wait I remember locking wasn't part of the malloc wrapper, it was separate therefore not implemented in `:platform:heap`: I think you need these retarget locks?
> https://gist.github.com/thomask77/65591d78070ace68885d3fd05cdebe3a

Thanks, will have a look into it.

mikewolfram, 2020-09-16:
Hmm, the retarget locks don't seem to work. Made a simple approach and get an assert. During static init everything is fine, but when the scheduler is not in right state (suspended) later on, it fails.

But tried a different way. vPortMalloc() uses vTaskSuspendAll() and xTaskResumeAll(). So I added it at the right places in heap_tlsf.c and it seems to work. Needs further/longer testing. If that really resolves it, then I would open a PR and modify the file to add the locking if module FreeRTOS is enabled.

salkinium, 2020-09-16:
> Hmm, the retarget locks don't seem to work.

Could be an issue of Newlib vs Newlib Nano, which is smaller, but has some limitations, like perhaps not using the malloc locks.
The specs are set here: https://github.com/modm-io/modm/blob/develop/ext/gcc/module_c.lb#L58

Unfortunately Newlib is criminally under-documented and short of looking at [the raw source code](https://github.com/mirror/newlib-cygwin), there's not much more I can help.

mikewolfram, 2020-09-16:
I created a static mutex in heap_tlsf.c and that fails. Didn't try newlib, but expect the same issue. Looking into your link he uses vTaskEnterCritical() and vTaskExitCritical() for malloc and free. Looking into vPortMalloc() is saw the suspend and resume calls.

When locking the mutex, xSemaphoreTake() is checking the task status and then it fails, but only after main, static init before is fine. Don't ask me.

salkinium, 2020-09-16:
There's also [this option in the FreeRTOS config file](https://github.com/modm-io/modm/blob/develop/ext/aws/FreeRTOSConfig.h.in#L58), that may need to be set to `1` here:

```c
#define configUSE_NEWLIB_REENTRANT              0
```

salkinium, 2020-09-16:
Ah, I'm sorry, we're using `heap_3.c`, which makes the calls from FreeRTOS threadsafe by wrapping the system provided malloc/free functions.

But I really didn't think this through: All calls from FreeRTOS are thread-safe, but malloc/free itself is not. So every plain C `malloc` or C++ `new` is not thread-safe. 🤦

salkinium, 2020-09-16:
Now I actually understand what you wrote!!! 🤦 Such an idiot I am…

salkinium, 2020-09-18:
I added thread-safe malloc via https://github.com/modm-io/modm/commit/cb82eecc054b8607e80b337af7a03b4b9a6abb09

mikewolfram, 2020-09-18:
Thanks. Have a few days off, will test next week.

mikewolfram, 2020-09-26:
Well, that thread-safe malloc/free didn't resolve the problem, still getting hard faults. I'm now suspecting the IPTask of FreeRTOS using more stack than they estimated.

salkinium, 2020-09-26:
FYI: You can use the [`:platform:fault` module](https://modm.io/reference/module/modm-platform-fault/) to report a memory dump for post-mortem debugging. That way you don't need a debugger connected all the time.

mikewolfram, 2020-10-25:
Haven't been here for quite some time, short update. Turned out, when running without any debugger attached, there are no memory problems. Had it running several days, with an open connection to our SECS host.

But when I run the app in debugger (either STLink or OpenOCD) the app will crash after some time. STLink sooner, OpenOCD sooner or later.

So where to go from here? Finish the implementation and have more people test it?

salkinium, 2020-10-25:
We can't help much without some code… I think @strongly-typed is quite interested in Ethernet support in modm.

closed 2021-09-29 by salkinium