modm-io/modm #449: FreeRTOS CPU clock set up

mikewolfram, 2020-08-24:
I'm experiencing a strange behavior when using FreeRTOS.
In FreeRTOSConfig.h I find this line:
```cpp
#define configCPU_CLOCK_HZ                      ( _ZN4modm5clock4fcpuE )
```
Having a LED blink at 1s interval let's it blink actually every 6-7 seconds. When I change the line to a constant of the real frequency, e.g.:
```cpp
#define configCPU_CLOCK_HZ                      ( 216000000 )
```
then the LED really blinks at 1s.

I checked modm::platform::fcpu, which is 216000000. Is there anything going on before Board::initialize() is called, which sets the frequency to modm::platform::fcpu? It is initialized with 16MHz first.

salkinium, 2020-08-25:
Ah, this hack again. `_ZN4modm5clock4fcpuE`  is `modm::clock::fcpu`, which is the wrong name, I'm surprised this even links at all. Try this with `_ZN4modm8platform4fcpuE`.

mikewolfram, 2020-08-25:
You are right, there's no modm::clock::fcpu, but it links without problems. I wonder which value it reads. :-)

closed 2020-08-26 by salkinium