modm-io/modm #468: Heap usage with Ubuntus packaged ARM Toolchain

rleh, 2020-09-15:
### Expected (with Toolchain [downloaded from ARM](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads))

```bash
user@vm$ which arm-none-eabi-gcc
/opt/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc
user@vm$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

user@vm$ cd ~/modm2/examples/nucleo_f411re/blink/
user@vm$ lbuild build
[...]
user@vm:$ scons -j4
[...]
Create Library· /home/user/modm2/build/nucleo_f411re/blink/release/modm/libmodm.a
Indexing······· /home/user/modm2/build/nucleo_f411re/blink/release/modm/libmodm.a
Linking········ /home/user/modm2/build/nucleo_f411re/blink/release/blink.elf
Memory usage··· /home/user/modm2/build/nucleo_f411re/blink/release/blink.elf
Program:   4.0 KiB (0.8% used)
(.build_id + .data + .fastcode + .fastdata + .hardware_init + .rodata +
 .table.copy.intern + .table.heap + .table.zero.intern + .text + .vector_rom)

Data:      5.2 KiB (4.1% used) = 2256 B static (1.7%) + 3072 B stack (2.3%)
(.bss + .data + .fastdata + .noinit + .stack)

Heap:    122.8 KiB (95.9% available)
(.heap1)

scons: done building targets.
```

### Reality (with [Ubuntus Toolchain](https://packages.ubuntu.com/focal/gcc-arm-none-eabi))

```bash
user@vm$ uname -a
Linux vm 5.4.0-47-generic #51-Ubuntu SMP Fri Sep 4 19:50:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
user@vm$ which arm-none-eabi-gcc
/usr/bin/arm-none-eabi-gcc
user@vm$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

user@vm$ cd ~/modm/examples/nucleo_f411re/blink/
user@vm$ lbuild build
[...]
user@vm$ scons -j4
[...]
Create Library· /home/user/modm/build/nucleo_f411re/blink/release/modm/libmodm.a
Indexing······· /home/user/modm/build/nucleo_f411re/blink/release/modm/libmodm.a
Linking········ /home/user/modm/build/nucleo_f411re/blink/release/blink.elf
`_sbrk_r' referenced in section `.text' of /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(lib_a-nano-mallocr.o): defined in discarded section `.Heap_is_not_implemented!__Please_include_the__modm:platform:heap__module_in_your_project!' of /home/user/modm/build/nucleo_f411re/blink/release/modm/libmodm.a(no_heap.o)
`_sbrk_r' referenced in section `.text' of /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(lib_a-nano-mallocr.o): defined in discarded section `.Heap_is_not_implemented!__Please_include_the__modm:platform:heap__module_in_your_project!' of /home/user/modm/build/nucleo_f411re/blink/release/modm/libmodm.a(no_heap.o)
`_sbrk_r' referenced in section `.text' of /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc_nano.a(lib_a-nano-mallocr.o): defined in discarded section `.Heap_is_not_implemented!__Please_include_the__modm:platform:heap__module_in_your_project!' of /home/user/modm/build/nucleo_f411re/blink/release/modm/libmodm.a(no_heap.o)
collect2: error: ld returned 1 exit status
scons: *** [/home/user/modm/build/nucleo_f411re/blink/release/blink.elf] Error 1
scons: building terminated because of errors.
```

After enabling `<module>modm:build:scons</module>`:
```bash
user@vm$ gedit project.xml
user@vm$ lbuild build
[...]
user@vm$ scons -j4
[...]
Create Library· /home/user/modm/build/nucleo_f411re/blink/release/modm/libmodm.a
Indexing······· /home/user/modm/build/nucleo_f411re/blink/release/modm/libmodm.a
Linking········ /home/user/modm/build/nucleo_f411re/blink/release/blink.elf
Memory usage··· /home/user/modm/build/nucleo_f411re/blink/release/blink.elf
Program:   5.0 KiB (1.0% used)
(.build_id + .data + .fastcode + .fastdata + .hardware_init + .rodata +
 .table.copy.intern + .table.heap + .table.zero.intern + .text + .vector_rom)

Data:      5.3 KiB (4.1% used) = 2316 B static (1.8%) + 3072 B stack (2.3%)
(.bss + .data + .fastdata + .stack)

Heap:    122.7 KiB (95.9% available)
(.heap1)

scons: done building targets.
```

### My fedora workstation with [arm-none-eabi-gcc-cs](https://bodhi.fedoraproject.org/updates/?packages=arm-none-eabi-gcc-cs)
Works without any problems, and uses less static memory compared to ARMs GCC?!
```bash
rleh@fedora$ which arm-none-eabi-gcc
/usr/bin/arm-none-eabi-gcc
rleh@fedora$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Fedora 9.2.0-4.fc32) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

rleh@fedora$ cd /examples/nucleo_f411re/blink/
rleh@fedora$ lbuild build
[...]
rleh@fedora$ scons -j4
[...]
Create Library· .../modm/build/nucleo_f411re/blink/release/modm/libmodm.a
Indexing······· .../modm/build/nucleo_f411re/blink/release/modm/libmodm.a
Linking········ .../modm/build/nucleo_f411re/blink/release/blink.elf
Memory usage··· .../modm/build/nucleo_f411re/blink/release/blink.elf
Program:   4.0 KiB (0.8% used)
(.build_id + .data + .fastcode + .fastdata + .hardware_init + .rodata +
 .table.copy.intern + .table.heap + .table.zero.intern + .text + .vector_rom)

Data:      5.2 KiB (4.1% used) = 2244 B static (1.7%) + 3072 B stack (2.3%)
(.bss + .data + .fastdata + .stack)

Heap:    122.8 KiB (95.9% available)
(.heap1)

scons: done building targets.
```

rleh, 2020-09-15:
<details>

<summary>

`arm-none-eabi-objdump -d blink.elf` (ARM Toolchain)

</summary>

```

/home/user/modm2/build/nucleo_f411re/blink/release/blink.elf:     file format elf32-littlearm

Disassembly of section .text:

080001a0 <_ZN4modm8IOStream4endlEv>:
 80001a0:	2300      	movs	r3, #0
 80001a2:	b510      	push	{r4, lr}
 80001a4:	4604      	mov	r4, r0
 80001a6:	6043      	str	r3, [r0, #4]
 80001a8:	6800      	ldr	r0, [r0, #0]
 80001aa:	6803      	ldr	r3, [r0, #0]
 80001ac:	210a      	movs	r1, #10
 80001ae:	689b      	ldr	r3, [r3, #8]
 80001b0:	4798      	blx	r3
 80001b2:	4620      	mov	r0, r4
 80001b4:	bd10      	pop	{r4, pc}

080001b6 <_ZN4modm8IOStreamlsEPKc>:
 80001b6:	b510      	push	{r4, lr}
 80001b8:	4604      	mov	r4, r0
 80001ba:	6800      	ldr	r0, [r0, #0]
 80001bc:	6803      	ldr	r3, [r0, #0]
 80001be:	68db      	ldr	r3, [r3, #12]
 80001c0:	4798      	blx	r3
 80001c2:	4620      	mov	r0, r4
 80001c4:	bd10      	pop	{r4, pc}
	...

080001c8 <_ZN5Board11SystemClock6enableEv>:
 80001c8:	b507      	push	{r0, r1, r2, lr}
 80001ca:	f44f 6000 	mov.w	r0, #2048	; 0x800
 80001ce:	f000 fcb5 	bl	8000b3c <_ZN4modm8platform3Rcc19enableInternalClockEm>
 80001d2:	4b1d      	ldr	r3, [pc, #116]	; (8000248 <_ZN5Board11SystemClock6enableEv+0x80>)
 80001d4:	6818      	ldr	r0, [r3, #0]
 80001d6:	9000      	str	r0, [sp, #0]
 80001d8:	889b      	ldrh	r3, [r3, #4]
 80001da:	f8ad 3004 	strh.w	r3, [sp, #4]
 80001de:	f44f 6200 	mov.w	r2, #2048	; 0x800
 80001e2:	4669      	mov	r1, sp
 80001e4:	2000      	movs	r0, #0
 80001e6:	f000 fcbd 	bl	8000b64 <_ZN4modm8platform3Rcc9enablePllENS1_9PllSourceERKNS1_10PllFactorsEm>
 80001ea:	4a18      	ldr	r2, [pc, #96]	; (800024c <_ZN5Board11SystemClock6enableEv+0x84>)
 80001ec:	6813      	ldr	r3, [r2, #0]
 80001ee:	f023 030f 	bic.w	r3, r3, #15
 80001f2:	f443 63e0 	orr.w	r3, r3, #1792	; 0x700
 80001f6:	f043 0303 	orr.w	r3, r3, #3
 80001fa:	6013      	str	r3, [r2, #0]
 80001fc:	f3bf 8f4f 	dsb	sy
 8000200:	f3bf 8f6f 	isb	sy
 8000204:	f44f 6100 	mov.w	r1, #2048	; 0x800
 8000208:	2002      	movs	r0, #2
 800020a:	f000 fcdb 	bl	8000bc4 <_ZN4modm8platform3Rcc17enableSystemClockENS1_17SystemClockSourceEm>
 800020e:	4b10      	ldr	r3, [pc, #64]	; (8000250 <_ZN5Board11SystemClock6enableEv+0x88>)
 8000210:	689a      	ldr	r2, [r3, #8]
 8000212:	f022 02f0 	bic.w	r2, r2, #240	; 0xf0
 8000216:	609a      	str	r2, [r3, #8]
 8000218:	689a      	ldr	r2, [r3, #8]
 800021a:	f422 52e0 	bic.w	r2, r2, #7168	; 0x1c00
 800021e:	f442 5280 	orr.w	r2, r2, #4096	; 0x1000
 8000222:	609a      	str	r2, [r3, #8]
[... 3395 lines cut ...]
 8001290:	6818      	ldr	r0, [r3, #0]
 8001292:	f7ff bf23 	b.w	80010dc <__wrap__free_r>
 8001296:	bf00      	nop
 8001298:	20000c08 	.word	0x20000c08

Disassembly of section .fastcode:

08001350 <_ZN4modm8delay_usEm>:
 8001350:	b148      	cbz	r0, 8001366 <_ZN4modm8delay_usEm+0x16>
 8001352:	4a05      	ldr	r2, [pc, #20]	; (8001368 <_ZN4modm8delay_usEm+0x18>)
 8001354:	4b05      	ldr	r3, [pc, #20]	; (800136c <_ZN4modm8delay_usEm+0x1c>)
 8001356:	8812      	ldrh	r2, [r2, #0]
 8001358:	6859      	ldr	r1, [r3, #4]
 800135a:	4350      	muls	r0, r2
 800135c:	3819      	subs	r0, #25
 800135e:	685a      	ldr	r2, [r3, #4]
 8001360:	1a52      	subs	r2, r2, r1
 8001362:	4282      	cmp	r2, r0
 8001364:	dbfb      	blt.n	800135e <_ZN4modm8delay_usEm+0xe>
 8001366:	4770      	bx	lr
 8001368:	20001506 	.word	0x20001506
 800136c:	e0001000 	.word	0xe0001000
```

</details>

salkinium, 2020-09-15:
What do you mean with `After enabling <module>modm:build:scons</module>:`? The Ubuntu Toolchain starts working if you add this module? Or do you have to call scons twice?

rleh, 2020-09-15:
> The Ubuntu Toolchain starts working if you add this module?

Yes. I added `<module>modm:build:scons</module>` to the `project.xml` file, then ran `lbuild build` and `scons -j4` again.

salkinium, 2020-09-15:
> I added <module>modm:build:scons</module> to the project.xml file, then ran lbuild build and scons -j4 again.

EXCUSE ME WAT?

I'm also seeing some other weird things like the TLSF implementation being part of *both* listings, even though that module only gets pulled in via `:platform:heap`? Something weird is going on here in modm, let me try to reproduce.

rleh, 2020-09-15:
Sorry, if miss-copy-pasted `<module>modm:platform:heap</module>` instead of `<module>modm:platform:heap</module>` in my issue and comments.
Of course I enabled the heap module.

rleh, 2020-09-15:
> I'm also seeing some other weird things like the TLSF implementation being part of both listings, even though that module only gets pulled in via :platform:heap?

Both objdumps are from from nucleo_f411re/blink example with added <module>modm:platform:heap</module> (for better comparison).

salkinium, 2020-09-15:
Can you give me the listing without the heap module? I want to see if anything pulls in srbk.

rleh, 2020-09-15:
> Can you give me the listing without the heap module?

I updated my comment above and added a listing from the ARM toolchain without heap module enabled.

> I want to see if anything pulls in srbk.

Me too, but I did not find it.

salkinium, 2020-09-15:
The two listings are identical, except for the non-elision of the fastcode section *name*. `modm::delay_us` is placed in `.fastcode` but on F411 this section is placed in Flash, due to the (small) cache.

So I still believe this is some weird linker issue. I'm not sure how to debug the linker at all.

```diff
< /home/user/modm2/build/nucleo_f411re/blink/release/blink.elf:     file format elf32-littlearm
---
> ../../../build/nucleo_f411re/blink/release/blink.elf:     file format elf32-littlearm
260c260
<  8000450:	f000 fd7a 	bl	8000f48 <__fastcode_load>
---
>  8000450:	f000 fd7a 	bl	8000f48 <_ZN4modm8delay_usEm>
400c400
<  8000592:	f000 fcd9 	bl	8000f48 <__fastcode_load>
---
>  8000592:	f000 fcd9 	bl	8000f48 <_ZN4modm8delay_usEm>
404c404
<  800059e:	f000 fcd3 	bl	8000f48 <__fastcode_load>
---
>  800059e:	f000 fcd3 	bl	8000f48 <_ZN4modm8delay_usEm>
1380,1381c1380,1381
<  8000f4a:	4a05      	ldr	r2, [pc, #20]	; (8000f60 <__fastcode_load+0x18>)
<  8000f4c:	4b05      	ldr	r3, [pc, #20]	; (8000f64 <__fastcode_load+0x1c>)
---
>  8000f4a:	4a05      	ldr	r2, [pc, #20]	; (8000f60 <_ZN4modm8delay_usEm+0x18>)
>  8000f4c:	4b05      	ldr	r3, [pc, #20]	; (8000f64 <_ZN4modm8delay_usEm+0x1c>)
```

rleh, 2023-03-12:
This issues was related to the GNU toolchain with GCC 9, therefore probably no longer relevant.

closed 2023-03-12 by rleh