modm-io/modm #469: Remapped pins on STM32F0/G0 not supported

chris-durand, 2020-09-15:
I have a board with an STM32F042F6 in a 20 Pin TSSOP package and would like to be able to use the remapped pins PA11/PA12. Some F0/G0 contain an internal mux which allows to select whether PA9/PA10 or PA11/PA12 are connected to the package pins. The F0 remaps both in a group whereas they can be mapped individually with the G0. The GPIO files for the remapped pins are currently not generated and the information is [currently not available](https://github.com/modm-io/modm-devices/issues/14) in modm-devices. The develop branch has the package pins pin, but it doesn't work with the with the current modm develop.

@salkinium What would it take to make this information available in modm? I would like to add support for remapped pins.

salkinium, 2020-09-15:
I did actually add this information, I just forgot to update the issue and close it: https://github.com/modm-io/modm-devices/issues/14#issuecomment-692974823

What's needed in modm is probably only updating the modm-devices submodule. The `:platform:gpio` module should just generate the additional pins.

There's currently no API for switching the pins, you have to do that manually. But otherwise there should be no limitations.

salkinium, 2020-09-15:
You'll probably have to fix a couple of unrelated issues due to more devices being added to modm. I can help with that if you don't feel up for it.

chris-durand, 2020-09-15:
I have already found this information but when I update the submodule to the current develop lbuild gives me:
```
ext/modm/repo.lb", line 93, in parse_all
    parser = modm_devices.parser.DeviceParser()
NameError: name 'modm_devices' is not defined
```

salkinium, 2020-09-15:
Ah sorry, I moved the implementation to top-level for PyPi, you need to change this at the top of `repo.lb`:
```diff
- sys.path.append(repopath("ext/modm-devices/tools/device"))
+ sys.path.append(repopath("ext/modm-devices"))
```

chris-durand, 2020-09-15:
Thanks, worked. Now modm also generates the `gpio_A11.hpp` and `gpio_A12.hpp` platform headers.

The question is how the interface for this should look like. The F0 remaps boths pins in a group but the normal modm gpio api does not model this; e.g. you could try to use A9 and A12 at the same time with different periperals which the hardware does not permit. With the normal peripheral connect methods there would be no option to warn the user at compile time. The options I see are to ignore it or to make it an lbuild option and only generate one set of A9/10 or A11/12 driver headers.

salkinium, 2020-09-15:
> The options I see are to ignore it 

I'd just ignore it for now, since I don't think the risk of accidentally using the wrong pin is high, since the datasheets mark these pins specifically.

There is no technical solution at compile time (would require keeping a program-wide state at compile time, not possible) anyways.

salkinium, 2020-09-15:
Do you want to open a PR with your working changes?

chris-durand, 2020-09-15:
I'm busy with something else right now, I'll do it in about 20min.

chris-durand, 2020-09-15:
The package pin data in modm device has string names but the gpios are described by number and port. So I have to transform pin 0, port f to `PF0` and match the strings to find out whether an io is a remap pin?

salkinium, 2020-09-15:
Yes, this is to reduce file size, since the information can be derived. (See https://github.com/modm-io/modm-devices/pull/46#issue-475804261)
```py
def pin_name(name):
    name = name[:4]
    if len(name) > 3 and not name[3].isdigit():
        name = name[:3]
    return (name[1:2].lower(), name[2:].lower())
```

chris-durand, 2020-09-15:
Sadly, I won't have time today and will open a PR tomorrow.

salkinium, 2020-09-15:
Ok, no worries, just didn't want this to get lost.

closed 2020-09-17 by salkinium