modm-io/modm #472: CAN init fails due to bus errors

mikewolfram, 2020-09-16:
During init of the CAN there is an assert, that it really initialized and the CAN status is ok.
```c++
	// Request leave initialization
	CAN1->MCR &= ~(uint32_t)CAN_MCR_INRQ;
	deadlockPreventer = 10'000; // max ~10ms
	while (((CAN1->MSR & CAN_MSR_INAK) == CAN_MSR_INAK) and (deadlockPreventer-- > 0))  {
		// wait for the normal mode
	}
	return modm_assert_continue_fail_debug(deadlockPreventer > 0, "can.init2",
		"CAN::initialize() timed out on leaving initialization!", 1);
```
In debug builds this causes the app to to abandon. It would be better to simply return a false also in debug builds.

The normal mode is only entered on a defined state of the RX line. In our case we use a TI ISO1042 isolated CAN transceiver, which might be without power if not needed. In such a case the STM32 won't start as it fails to init the CAN interface. Unfortunately the ISO1042 affects the RX signal when powered off, some boards are fine, some fail.

I've also seen issues with CAN termination on long cables causing the STM32 to fail here.

salkinium, 2020-09-16:
From the `max ~10ms` comment, I suspect that while loop is missing a `modm::delay_us(1)` inside, thus in slightly slower debug mode can fail.

However, I would agree with you that it can be better to return a boolean here.
cc @strongly-typed Opinions?

Alternatively you can selectively ignore this assertion only in debug mode:

```cpp
static modm::Abandonment
ignore_can_init(const modm::AssertionInfo &info)
{
    if (info.name == "can.init2"s)
        return Abandonment::Ignore;
    return Abandonment::DontCare;
}
MODM_ASSERTION_HANDLER_DEBUG(ignore_can_init);
```

salkinium, 2020-09-16:
See also https://modm.io/reference/module/modm-architecture-assert/#assertion-handlers for additonal ways to ignore such assertions.

salkinium, 2020-09-16:
Maybe changing to `modm_assert_continue_fail_ignore` (resuming unless overwritten) would be acceptable?

mikewolfram, 2020-09-16:
If you ask me, I'd simply return a false in that case letting the user handle it.

This leads me to another question. If the init fails here because of bus error and the bus error disappears a bit later, would the STM32 itself finish the return to normal? Or would it require a new init? Also the init might be ok, but a bus error shows up later.

So I propose to simply return a false to indicate the CAN isn't in normal operation and the users app is in charge to retry later.

And yes, adding a 1us delay could help to wait the full 10ms. :-) Which leads me to another question, will open a new issue.

strongly-typed, 2020-09-17:
> However, I would agree with you that it can be better to only return a boolean here.
> cc @strongly-typed Opinions

> If you ask me, I'd simply return a false in that case letting the user handle it.

There is no way to enforce (at least to my knowledge, please prove me wrong) for the user to handle it. At @roboterclubaachen, an up and running CAN is literally mission critical and so we decided that blocking would be better than passing over it. If the user does not handle the error on start-up, we at least see by the missing blink of LEDs of main that something was wrong. That's of course a very crude, but practical approach to a practical problem.

I could live with a bool return. Maybe add something like `init_blocking` for the old behavior, if required? cc @rleh

rleh, 2020-09-17:
> I could live with a bool return. Maybe add something like init_blocking for the old behavior, if required? cc @rleh

In my opinion returning a (`[[nodiscard]]`) bool is fine. I don't think an `init_blocking()` is useful, `while(1) {}` is better done in the application code if there is no better solution.

mikewolfram, 2020-09-26:
Tried to add the `[[nodiscard]]` attribute to the function, but my gcc gives me a warning, that the attribute is ignored.

Using
`gcc version 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599] (xPack GNU Arm Embedded GCC, 64-bit)`

So my assumption is, that it cannot be used on class members, even when static.

Go without `[[nodiscard]]` attribute?

mikewolfram, 2020-10-07:
Raising this again. Any opinions?

salkinium, 2020-10-07:
Just open a PR with `[[nodiscard]]` and we'll discuss further from there.

PR #491 (2020-10-10): Avoid fail of CAN init when the bus isn't ready.
Return a bool instead and let the user handle the situation.

Fixes #472.
closed 2020-10-10 by salkinium