modm-io/modm #473: modm::delay and FreeRTOS

mikewolfram, 2020-09-16:
Discovered something interesting: In a FreeRTOS task, when using modm::delay (any variant) the task is blocked and also preventing the other tasks to execute. I would expect, that the delay function would allow other tasks to run.

Observed it first when used a delay in a task and the IP task never got a chance to run.

Pre-emptive is turned on.

salkinium, 2020-09-16:
Uff, that's very bad!

The delay functions are [**polling** the free-running DWT Cycle Counter](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/delay.cpp.in), it doesn't seem obvious to me how this would block anything really.

salkinium, 2020-09-16:
Could it be that FreeRTOS depriviledges tasks by default since it wants to use the MPU for stack overflow protection? In unpriviledged mode you cannot access the DWT, which would cause a Usage(?) Fault if I recall correctly. But that's remapped to HardFault in modm.

salkinium, 2020-09-16:
Uuiuiui, I think it does depriviledge all tasks, see this option:

```c
#define configINCLUDE_APPLICATION_DEFINED_PRIVILEGED_FUNCTIONS 0
```

https://www.freertos.org/a00110.html#configINCLUDE_APPLICATION_DEFINED_PRIVILEGED_FUNCTIONS

Ok, so the quickfix is to fallback to instruction counting when also using the `env.has_module(":freertos")` (code already exists, just needs an extra bool), which is less accurate.

mikewolfram, 2020-09-16:
This is the implementation:
```c+++
	asm volatile (
		".syntax unified"       "\n\t"
		"muls.n	%0, %0, %1"     "\n\t"  // get number of cycles by us * fcpu_MHz:               1-2 cycles on cm3, up to 32 cycles on cm0
	"1:  subs.n	%0, %0, #1"     "\n\t"  // subtract the loop cycles from the input:             1 cycle
		"bpl.n	1b"             "\n\t"  // keep doing that while result is still positive:      2 cycles (when taken)
	:: "r" (us), "r" (platform::delay_fcpu_MHz));
```
So no DWT.

mikewolfram, 2020-09-16:
But still I don't see why it would prevent to switch tasks. Preemptive means, the tick IRQ would switch the task. Or am I wrong?

salkinium, 2020-09-16:
No, this is also what I would expect and especially not when using the instruction counter method, which doesn't have anything to do with interrupts or interrupt priorities or locking or whatever else could be blocking it.

salkinium, 2020-09-17:
You can replace the delay implementation with a while() loop in C, the algorithm is very simple. That should remove any doubt regarding the implementation.

salkinium, 2021-01-10:
Did you find a solution to this?

mikewolfram, 2021-01-11:
Not yet, I was too busy to get the project up and running before Xmas. The next 2-3 months I'll spent with clean-up, some restructuring and will also fix whatever was discovered during development.

salkinium, 2021-06-16:
I changed the delay implementation, so maybe this is fixed now. I'm closing this for now.

closed 2021-06-16 by salkinium