modm-io/modm #480: How to use MODM_FLAGS with a single HW access method?

strongly-typed, 2020-09-19:
I read [the blog](https://blog.salkinium.com/typesafe-register-access-in-c++/) and [documentation](https://raw.githubusercontent.com/modm-io/modm/develop/src/modm/architecture/interface/register.md).

Assume multiple different registers (a register map from some extensive peripheral), e.g.:
```c++
enum class
ECON2 : uint8_t {
	AUTOINC = modm::Bit7,
	PKTDEC  = modm::Bit6,
};
MODM_FLAGS8(ECON2);

enum class
ECON1 : uint8_t {
	TXRST = modm::Bit7,
	RXRST = modm::Bit6,
};
MODM_FLAGS8(ECON1);

...
...
```

```C++
enum class Register : uint8_t
{
	ECON1 = 0x1F,
	ECON2 = 0x1E,
	...,
	...,
}
```

and a single HW access method:
```C++
static void
set(Register reg, uint8_t address);
```

How can `set(Register::ECON1, ECON1::TXRST);` and `set(Register::ECON2, ECON2:: AUTOINC);` aliased to `set(Register reg, uint8_t address);` without manually doing something like this for all registers?

```C++
static void set(Register reg, ECON1_t econ1) { set(reg, econ1.value); }
static void set(Register reg, ECON2_t econ2) { set(reg, econ2.value); }
...
...
```

Or is there any suitable approach in `modm`? Any thoughts appreciated!

chris-durand, 2020-09-19:
Have a look at `modm::FlagsGroup`, for example used [here](https://github.com/modm-io/modm/blob/6e5ebf4b63932ed4b137d9697576546ea41d3a75/src/modm/driver/pwm/lp503x.hpp.in#L81).

salkinium, 2020-09-19:
Yep, `modm::FlagsGroup` is the solution.

strongly-typed, 2020-09-19:
Great, but missed in the documentation?

salkinium, 2020-09-19:
> Great, but missed in the documentation?

That's very possible, the docs need to be updated.

henrikssn, 2020-09-20:
You can technically also use `modm::Register8` (it isn't clear to me what the difference w.r.t `modm::FlagsGroup` is)

salkinium, 2020-09-20:
`modm::Register8` will allow any flags to be put in there, while `modm::FlagsGroup` only allows the ones from the group. This is useful when you only want to allow *some* flags in your device to be implicitly castable into your function.

henrikssn, 2020-09-20:
Cool, I did not know that!

*Erik busy with changing his code to use `modm::FlagsGroup`*

closed 2021-06-16 by salkinium