modm-io/modm #483: Ili9341 SPI not compiling

mikewolfram, 2020-09-26:
The Ili9341 doesn't compile for me anymore when using with SPI interface. In extension to my implementation #464 the code in the #437 adds a parallel interface. But the templated constructor prevents from compiling. Also for SPI it expects an argument to the constructor.

Maybe you could have a look into it?

salkinium, 2020-09-26:
cc @ASMfreaK

ASMfreaK, 2020-09-26:
I'll look into it today.

ASMfreaK, 2020-09-26:
It seems, that I've made a mistake in `stdL enable`.

rleh, 2020-09-27:
It would be nice to have examples for Ili9341 both with SPI and parallel mode. Thus the error would be noticeable in the CI in the future.

ASMfreaK, 2020-09-27:
Yeah. That were my thought exactly yesterday. I've created an example to. Will try to fix and add a new example.

PR #484 (2020-09-27): [driver] :bug: Fix ILI9341 and add SPI example for it
This closes #483 and adds an example for ILI9341 in SPI mode for Blue Pill.
Example compiles. It **should** work on real hw, but I cannot confirm this.
closed 2020-09-27 by salkinium