modm-io/modm #492: Python tools fail with cmake: "cannot import name 'SkipTest' from 'unittest'"

mcbridejc, 2020-10-10:
When executing `python3 modm/modm_tools/openocd.py -f modm/openocd.cfg program.elf`, I get the following error:

```
Traceback (most recent call last):
  File "modm/modm_tools/openocd.py", line 33, in <module>
    from modm_tools import utils, backend as bem
  File "modm/modm_tools/__init__.py", line 25, in <module>
    from . import info
  File "modm/modm_tools/info.py", line 20, in <module>
    from git import Repo
  File "/home/jeff/.local/lib/python3.8/site-packages/git/__init__.py", line 40, in <module>
    from git.config import GitConfigParser  # @NoMove @IgnorePep8
  File "/home/jeff/.local/lib/python3.8/site-packages/git/config.py", line 24, in <module>
    from git.util import LockFile
  File "/home/jeff/.local/lib/python3.8/site-packages/git/util.py", line 18, in <module>
    from unittest import SkipTest
ImportError: cannot import name 'SkipTest' from 'unittest' (/home/jeff/dev/touchdemo-c++/modm/modm_tools/unittest.py)
```

The problem seems clear: `sys.path[0]` points to `<projectdir>/modm/modm_tools`, so the modm provided `unittest.py` is being imported instead of the python standard package. I don't see anything special about my system to cause this, except that I am using cmake. Using scons, `scons program` works fine, I believe because in this case the `openocd.py` is imported, not run as an executable, and therefor its path is not set in `sys.path[0]` by the python interpreter.

salkinium, 2020-10-10:
Whoops, that's unfortunate… If you renamed `unittest.py` to, say `unit_test.py`, does the problem go away?

salkinium, 2020-10-10:
This is probably also contributing to this issue. modm_tools isn't a real python package, it should probably not import the whole folder and overwrite the top-level modules this way. I'm not entirely sure what I was thinking when I did this?

```py
if __name__ == "__main__":
    import sys
    sys.path.append(os.path.dirname(os.path.dirname(__file__)))
```

mcbridejc, 2020-10-10:
> Whoops, that's unfortunate… If you renamed `unittest.py` to, say `unit_test.py`, does the problem go away?

It does.

mcbridejc, 2020-10-10:
> This is probably also contributing to this issue. modm_tools isn't a real python package, it should probably not import the whole folder and overwrite the top-level modules this way. I'm not entirely sure what I was thinking when I did this?
> 
> ```python
> if __name__ == "__main__":
>     import sys
>     sys.path.append(os.path.dirname(os.path.dirname(__file__)))
> ```

I thought this at first too, but then I realized this is taking the parent directory twice, so it is adding `modm` to the path, not `modm/modm_tools` where unittest is. I'm pretty sure this is why it gets imported, from https://docs.python.org/3/library/sys.html#sys.path: 

> As initialized upon program startup, the first item of this list, path[0], is the directory containing the script that was used to invoke the Python interpreter. If the script directory is not available (e.g. if the interpreter is invoked interactively or if the script is read from standard input), path[0] is the empty string, which directs Python to search modules in the current directory first.

salkinium, 2020-10-11:
>> Whoops, that's unfortunate… If you renamed unittest.py to, say unit_test.py, does the problem go away?
>
> It does.

Then let's just rename it. Do you want to open a PR?

mcbridejc, 2020-10-12:
Haha. This is a good one: 
[image]

closed 2020-10-12 by salkinium