modm-io/modm #501: ITM example does not work

strongly-typed, 2020-11-16:
I am currently not quite sure how to run the example in [examples/nucleo_f103rb/itm](https://github.com/modm-io/modm/tree/develop/examples/nucleo_f103rb/itm).

I slightly modified the example to run on a Blue Pill: [feature/itm](https://github.com/strongly-typed/modm/commit/bbeb364569b7985c12ad80bcdb694c0f05e2482b)

When I run (after lbuild and programming)

```
scons log-itm fcpu=64000000
```

I just get
```
➜  itm git:(feature/itm) ✗ scons log-itm fcpu=64000000
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
.----OpenOCD--> Single Wire Viewer
'------SWO----- stm32f103c8t6
Open On-Chip Debugger 0.10.0+dev-01514-ga8edbd020-dirty (2020-11-16-21:53)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
modm_program
```

and no further output. I tried with an OpenOCD version from January and the latest HEAD.

Any clues?

strongly-typed, 2020-11-16:
When I change the blink frequency to 5 Hz (delay 100ms), I notice a change of blink frequency when I run `scans log-itm`. Seems to me that the (hard) fault handler caught in. 

Good to have the same blink pattern in the blinky example and the hard fault handler. So whatever happens, the blinky example always works! modm **has** a marketing department!

salkinium, 2020-11-17:
ITM logs over the SWO pin, which isn't connected on the Blue-Pill?

We should probably implement [RTT logging](https://www.segger.com/products/debug-probes/j-link/technology/about-real-time-transfer/), which only requires a debug connection, and works asynchronously.

strongly-typed, 2020-11-17:
> ITM logs over the SWO pin, which isn't connected on the Blue-Pill?

OK, got it. I only connected `SWCLK` and `SWDIO` to a cheap ST-Link V2 clone. And my expectation was (without reading the doc) that this already bidirectional data connection is also used for logging.

With a cheap ST-Link V2 clone, some [HW mod is required](https://lujji.github.io/blog/stlink-clone-trace/). I'll look into that today.

strongly-typed, 2020-11-17:
Jupp, works!

```
➜  itm git:(feature/itm) ✗ scons log-itm fcpu=72000000
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
.----OpenOCD--> Single Wire Viewer
'------SWO----- stm32f103c8t6
Open On-Chip Debugger 0.10.0+dev-01514-ga8edbd020-dirty (2020-11-16-21:53)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
modm_program
loop: 1191
loop: 1192
loop: 1193
```

closed 2020-11-17 by strongly-typed