modm-io/modm #505: Is there an example that show how to use STM32 usb device?

zhangzq71, 2020-11-17:
I can't find any examples in modm examples dir that show how to use STM32 usb device, such as a CDC device.

salkinium, 2020-11-17:
Here are a couple:

- Generic Example for STM32 and SAM devices: MSC, CDC, MIDI: https://github.com/modm-io/modm/tree/develop/examples/generic/usb
- FIrmware upgrade via MSC and FatFs: https://github.com/modm-io/modm/tree/develop/examples/stm32f411ceu_mini_f411/usbfatfs
- Firmware upgrade via DFU runtime: https://github.com/modm-io/modm/tree/develop/examples/stm32f3_discovery/usb_dfu

salkinium, 2020-11-17:
The trick is to set the lbuild option `<option name="modm:tinyusb:config">device.cdc</option>` to generate the USB descriptors, and then use `modm::platform::UsbUart{#ITM}` as an IODevice.

```cpp
modm::IODeviceWrapper<modm::platform::UsbUart0, modm::IOBuffer::DiscardIfFull> usb_io_device0;
modm::IOStream usb_stream0(usb_io_device0);
```

zhangzq71, 2020-11-17:
Thank you very much, let me try.

closed 2020-11-17 by zhangzq71

salkinium, 2020-11-17:
I ran out of motivation for implementing more USB descriptors, so have a look at the TinyUSB examples as well, since modm is only a very light wrapper over the TinyUSB core code: https://github.com/hathach/tinyusb/tree/master/examples/device

zhangzq71, 2020-11-17:
Thanks, I knew modm one day before and feel it is great and learning how to use it. And the usb function I need is the CDC ACM, and there is an example for it, that is enough.