modm-io/modm #506: [boost] placeholders (_1, _2, ...) in the global namespace is deprecated

strongly-typed, 2020-11-22:
```
/usr/local/include/boost/config/pragma_message.hpp:24:34: note: '#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.'
   24 | # define BOOST_PRAGMA_MESSAGE(x) _Pragma(BOOST_STRINGIZE(message(x)))
      |                                  ^~~~~~~
/usr/local/include/boost/bind.hpp:36:1: note: in expansion of macro 'BOOST_PRAGMA_MESSAGE'
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/local/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/local/include/boost/shared_ptr.hpp:17,
                 from /usr/local/include/boost/date_time/time_clock.hpp:17,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:10,
                 from /usr/local/include/boost/asio/time_traits.hpp:23,
                 from /usr/local/include/boost/asio/detail/timer_queue_ptime.hpp:22,
                 from /usr/local/include/boost/asio/detail/deadline_timer_service.hpp:29,
                 from /usr/local/include/boost/asio/basic_deadline_timer.hpp:25,
                 from /usr/local/include/boost/asio.hpp:25,
                 from modm/src/modm/platform/uart/serial_port.hpp:20,
                 from modm/src/modm/platform/uart/serial_port.cpp:14:
```

We might run into future errors as it is deprecated.

rleh, 2020-11-22:
I don't see any usage of `placeholder::_1` (etc.) in modm.
In [serial_port.cpp](https://github.com/modm-io/modm/blob/develop/src/modm/platform/uart/hosted/serial_port.cpp) and [serial_port.hpp](https://github.com/modm-io/modm/blob/develop/src/modm/platform/uart/hosted/serial_port.hpp) we should be able to use C++ (since 17) [`std::bind`](https://en.cppreference.com/w/cpp/utility/functional/bind) and [`std::placeholders`](https://en.cppreference.com/w/cpp/utility/functional/placeholders) instead.

chris-durand, 2020-11-22:
The warning is issued when `boost/bind.hpp` is included which has `using namespace boost::placeholders` inside the header. Including `boost/bind/bind.hpp` is the recommended way to use `boost::bind`, if you have to. The modm driver is using `boost::asio::placeholders::…` which does not work with `std::bind`. The easiest way to resolve the deprecation warning is changing the include. You could use a lambda, though.

chris-durand, 2020-11-22:
Does anyone know why there are two distinct hosted implementations of a serial port? We have the asio-based `SerialPort` and hand-rolled, linux-only `SerialInterface`. Both implement the `modm::IODevice` interface.
`SerialInterface` is just a thin wrapper around syscalls. `SerialPort` performs all IO asynchronously in a background thread and manages its own variable size buffers.
Quickly looking at the code it was quite apparent that both implementations could profit from some refactoring. I am wondering if we could have just one, but I am not completely sure how this code is used.

@strongly-typed and others, how are you using that code?

strongly-typed, 2020-11-22:
> Does anyone know why there are two distinct hosted implementations of a serial port? 

10 years of xpcc history? :-)

> SerialInterface is just a thin wrapper around syscalls. SerialPort performs all IO asynchronously in a background thread

I think `SerialPort` came from @TheTh0r (0beb439f502b8a351e3231625b177de482ff0747 on 2010-08-14) implementation of CAN2USB (refactored from `SerialPort` to `SerialInterface` in 67f69b107cf1ced71678e80bfcfeebe56f1b66a9 by @salkinium (was there any hassle with boost?) and then later with a template in 9cc22bdd81b427b0e1de8d5c037936702927aaf4). I think `SerialPort` did not work correctly until 44f072f8541f0fa06e94c780b6948e47dd969e2d on 2010-08-17.

`SerialInterface` was added shortly after that in b710181cb970c2b488838df7b63080e7858f0c13 on 2010-11-29 by @fabian-dlr, credits in the commit are "Philipp & Metty".

I was just using `SerialInterface` to connect a good old CanUsb to my laptop and access a CAN bus. So in active use on MacOS ;-)

ekiwi, 2020-11-23:
I don't think anyone ever really figgured out how to use Boost ASIO correctly.

TheTh0r, 2020-11-23:
I remember, that at least the implementation worked fine. If I used Boost ASIO as intended I dont know. Today, it could most likely be written much cleaner using lambda functions :)

ASMfreaK, 2020-11-23:
I've used Serial Port and SerialInterface. I refactored it quite a bit for it to have the same behavior as modm::Uart on mcus. I can publish the code, but I doubt it'll fix this issue.

TheTh0r, 2020-11-23:
ASMfreaK, could you push this to a new branch? I can use the Christmas holidays to look at my programming sins and maybe update the SerialPort implementation to a c++14 style.

PR #511 (2020-11-25): [hosted] Fix include of boost bind in serial_port.hpp
Fixes #506
closed 2020-11-25 by salkinium