modm-io/modm #510: How to transfer 16bit SPI data?

zhangzq71, 2020-11-24:
I want to make a SPI encoder driver, I have to transfer 16bit SPI data, but I found the SpiMaster1 has only the 8bit transfer function.

`
modm::ResumableResult<uint8_t>
modm::platform::SpiMaster1::transfer(uint8_t data)
{
	// this is a manually implemented "fast resumable function"
	// there is no context or nesting protection, since we don't need it.
	// there are only two states encoded into 1 bit (LSB of state):
	//   1. waiting to start, and
	//   2. waiting to finish.

	// LSB != Bit0 ?
	if ( !(state & Bit0) )
	{
		// wait for previous transfer to finish
		if (!SpiHal1::isTransmitRegisterEmpty())
			return {modm::rf::Running};

		// start transfer by copying data into register
		SpiHal1::write(data);

		// set LSB = Bit0
		state |= Bit0;
	}

	if (!SpiHal1::isReceiveRegisterNotEmpty())
		return {modm::rf::Running};

	SpiHal1::read(data);

	// transfer finished
	state &= ~Bit0;
	return {modm::rf::Stop, data};
}
`
how can I transfer 16bit SPI data?

salkinium, 2020-11-24:
SpiHal1 exposes a 16-bit version of the `write` function, as well as `setDataWidth` if the hardware supports it. Unfortunately the interface is 8-bit only, so you have to use SpaHal directly for now, sorry!

zhangzq71, 2020-11-25:
Ok, thanks, let me try.

closed 2020-11-25 by zhangzq71