modm-io/modm #514: Can I use platform::flash in stm32f030c8t6?

zhangzq71, 2020-11-28:
I got this error when running 'lbuild build'
ERROR: Cannot resolve name 'modm:platform:flash' of type Module!

what is the problem?

salkinium, 2020-11-28:
The Module `modm:platform:flash` is only available for STM32F4 and STM32G0, because I was lazy and only implemented it for those two Flash peripherals:
https://github.com/modm-io/modm/blob/develop/src/modm/platform/flash/stm32/module.lb#L20

You will have to port this to STM32F0 or use raw register access.

zhangzq71, 2020-11-28:
Ok, thanks, I got it.

closed 2020-11-28 by zhangzq71

salkinium, 2020-11-28:
I compared the STM32F0 and STM32G0 Reference Manual sections on the Flash Peripheral, and they seem to work very similiarly with 1-2kB page size, only some registers bit fields have different names (`FLASH_SR_BSY1` vs `FLASH_SR_BSY`). 

So if you have time and motivation you could adapt the STM32G0 driver for the STM32F0 one.

zhangzq71, 2020-11-29:
oh? that sounds good, let me try. thanks.

zhangzq71, 2020-11-30:
it can't directly use STM32G0 driver for STM32F0, 
`
modm_ramcode uint32_t
Flash::erase(uint8_t index)
{
	FLASH->SR = FLASH_SR_ERR;
	FLASH->CR = FLASH_CR_STRT | FLASH_CR_PER |
			((index << FLASH_CR_PNB_Pos) & FLASH_CR_PNB_Msk);
	while(isBusy()) ;
	FLASH->CR = 0;

	return FLASH->SR & FLASH_SR_ERR;
}
`

STM32F0  FLASH_CR register doesn't have FLASH_CR_PNB_Pos bit.

salkinium, 2020-11-30:
Oh, that's sad! I guess the drivers are a little more different than I thought. I think I read something about accessing a memory location inside flash, and then the hardware automatically calculates the sector/page from that. But it's all inconsistent across STM32. I was a little overwhelmed by all the little differences like this, hence only implementing it for two families to at least get *something* started.

zhangzq71, 2020-11-30:
You've done most of the work, I think there is slight difference, I check the STM32 user manual to see if I can get the STM32F0 flash work.