modm-io/modm #515: Access to milli_clock hangs the test

strongly-typed, 2020-12-06:
I was curious to access `now` of `milli_clock` in testing.

[Unfortunately, the code just hangs](https://github.com/strongly-typed/modm/commit/b9f3ea379b0f97a66642fd8a20e5eddf9a2c5255) when I run `(cd test && make run-hosted-darwin)`.

Any idea?

Instead of just using the setters, I wanted to have a getter of the current simulated time. What would be the preferred way to do that?

salkinium, 2020-12-06:
The implementation of `milli_clock` is replaced by a mockup that allows manual time advances. Otherwise testing timeouts may not be reproducible or take way too much time: https://github.com/modm-io/modm/blob/develop/test/modm/mock/clock.hpp

There should be no reliance on any real-time clock in the unittests, unless there is a specific reason (like testing a hardware clock like MCO output on a Nucleo or something like that).

chris-durand, 2020-12-06:
> ```
> static inline modm::chrono::milli_clock::time_point now() noexcept
> { return now(); }
> ```

This will recursively call itself and thus, just hangs.

salkinium, 2020-12-06:
Why are you modifying the AMNB test for this? Is it not working correctly?

chris-durand, 2020-12-06:
@salkinium I don't understand why `modm_test::chrono::milli_clock` inherits privately from `modm::chrono::milli_clock`. It does not access anything inherited from the private base class. Did you intend to write `class milli_clock : public modm::chrono::milli_clock`, instead of `class milli_clock : modm::chrono::milli_clock`?

strongly-typed, 2020-12-06:
I was modifying AMNB just to see/demonstrate if/how it works and not introducing the whole communication stack code I am currently developing.

So then is my question: In a test I would like to advance the time by `increment(1)` repeatedly and then check if something changed. I would like to access the current time that elapsed. How would I do that?

chris-durand, 2020-12-06:
Call `modm_test::chrono::increment()`, and read the value returned by `modm::chrono::milli_clock::now()`. Both operate on a global variable `milli_time` in [`clock.cpp`](https://github.com/modm-io/modm/blob/develop/test/modm/mock/clock.cpp#L16).

salkinium, 2020-12-06:
> . It does not access anything inherited from the private base class. 

You're right, it doesn't need to inherit anything here. Brainfarts on my part.

strongly-typed, 2020-12-06:
> modm::chrono::milli_clock::now()

Interesting, I did exactly that in my project before adding the `now()` thing. In the AMNB node, it works.

chris-durand, 2020-12-06:
To me it looks like, public inheritance was intended here. Then `modm_test::chrono::milli_clock` would be a fully functional `milli_clock` with added `increment()` and `setTime()` member functions. If a thing is called `modm_test::chrono::milli_clock`, I would expect it to provide the functionality of `modm::chrono::milli_clock`. If it is only able to set the time it should be clear to the user from its name.

salkinium, 2020-12-06:
I don't remember what my intention was, or if there was any at all, I just wanted to have something working that didn't require refactoring all the tests. This meant replacing the `now()` implentations, since a lot of `modm::milli_clock` usage was hardcoded in the library. I only made the `now()` functions weak so that you can also overwrite the functions to use the RTC or other timers in future applications.

salkinium, 2020-12-06:
If you feel like working on this, I'll hapily merge any changes, but the modm unit testing code hasn't been shown a lot of interest, perhaps the introduction of the mock modules has made them a bit more useful lately, so we should perhaps spend some more time on those.

closed 2021-06-16 by salkinium