modm-io/modm #516: Can't compile STM32f4_discovery examples

ayoubSoussi, 2020-12-08:
Hi,
I'm trying to compile an example of the modm library to get started with it, but I found some errors during the build. I had the same errors in 2 different examples : stm32f4_discovery/blink and stm32f4_discovery/uart.

### Problem : 
Using the command
`make build`
I get the errors
[image]
  
Saying that __get_BASEPRI, __set_BASEPRI_MAX, __set_BASEPRI were not declared.

I think that the examples in this folder are out-of-date (if it's the case, I would like to get a hint to what to do and I can update the examples).

Thanks.

salkinium, 2020-12-08:
You need to do `make cmake` once, then you can do `make build`.

We should make the Makefile `cmake` target depend on the actual folders, so that the cmake part get's automatically build on `make build`.

ayoubSoussi, 2020-12-08:
of course I did a `make cmake`, then a 'make build', but I still get the error

rleh, 2020-12-08:
What operation system are you using?
Which GCC version (`arm-none-eabi-g++ --version`) do you use?

ayoubSoussi, 2020-12-08:
I use Ubuntu 18.04.5 LTS
My GCC version is : 9.3.1

## Update : 
I was able to compile and generate the bin and hex files using this combination of commands
`make cmake`
`make build ` (it still gives the error I described earlier)
`make all`

rleh, 2020-12-08:
I can't reproduce the error on my Fedora Linux system, but get another error:
```bash
...
arm-none-eabi-objdump: DWARF error: could not find variable specification at offset 20cd
arm-none-eabi-objdump: DWARF error: could not find variable specification at offset 964
arm-none-eabi-objdump: DWARF error: could not find variable specification at offset 973
arm-none-eabi-objdump: DWARF error: could not find variable specification at offset 1a0e
arm-none-eabi-objdump: DWARF error: could not find variable specification at offset 1a1d
...
```

On Ubuntu 20.04 is get the following error:
```bash
rleh@ubuntu-server-20:~/modm/examples/stm32f4_discovery/uart$ make cmake
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- Check for working C compiler: /home/rleh/.local/arm-none-eabi-gcc/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Check for working C compiler: /home/rleh/.local/arm-none-eabi-gcc/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/home/rleh/.local/arm-none-eabi-gcc/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/rleh/modm/build/stm32f4_discovery/uart/cmake-build-debug/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_19d4f/fast && make[1]: Entering directory '/home/rleh/modm/build/stm32f4_discovery/uart/cmake-build-debug/CMakeFiles/CMakeTmp'
    /usr/bin/make -f CMakeFiles/cmTC_19d4f.dir/build.make CMakeFiles/cmTC_19d4f.dir/build
    make[2]: Entering directory '/home/rleh/modm/build/stm32f4_discovery/uart/cmake-build-debug/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_19d4f.dir/testCCompiler.c.obj
    /home/rleh/.local/arm-none-eabi-gcc/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc   -fdata-sections     -ffunction-sections     -finline-limit=10000     -fshort-wchar     -fsingle-precision-constant     -funsigned-bitfields     -funsigned-char     -fwrapv     -g3     -gdwarf-3     -W     -Wall     -Wdouble-promotion     -Wduplicated-cond     -Werror=format     -Werror=maybe-uninitialized     -Werror=overflow     -Werror=sign-compare     -Wextra     -Wlogical-op     -Wno-redundant-decls     -Wpointer-arith     -Wundef     -mcpu=cortex-m4     -mfloat-abi=hard     -mfpu=fpv4-sp-d16     -mthumb      -std=gnu11     -Wbad-function-cast     -Wimplicit     -Wredundant-decls     -Wstrict-prototypes     -o CMakeFiles/cmTC_19d4f.dir/testCCompiler.c.obj   -c /home/rleh/modm/build/stm32f4_discovery/uart/cmake-build-debug/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C static library libcmTC_19d4f.a
    /usr/bin/cmake -P CMakeFiles/cmTC_19d4f.dir/cmake_clean_target.cmake
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_19d4f.dir/link.txt --verbose=1
    arm-none-eabi-ar qc libcmTC_19d4f.a  CMakeFiles/cmTC_19d4f.dir/testCCompiler.c.obj
    Error running link command: No such file or directory
    make[2]: *** [CMakeFiles/cmTC_19d4f.dir/build.make:88: libcmTC_19d4f.a] Error 2
    make[2]: Leaving directory '/home/rleh/modm/build/stm32f4_discovery/uart/cmake-build-debug/CMakeFiles/CMakeTmp'
    make[1]: *** [Makefile:121: cmTC_19d4f/fast] Error 2
    make[1]: Leaving directory '/home/rleh/modm/build/stm32f4_discovery/uart/cmake-build-debug/CMakeFiles/CMakeTmp'
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:15 (project)

-- Configuring incomplete, errors occurred!
See also "/home/rleh/modm/build/stm32f4_discovery/uart/cmake-build-debug/CMakeFiles/CMakeOutput.log".
See also "/home/rleh/modm/build/stm32f4_discovery/uart/cmake-build-debug/CMakeFiles/CMakeError.log".
make: *** [Makefile:25: cmake] Error 1
```

salkinium, 2020-12-08:
(On macOS it works fine, with arm-none-eabi-gcc 9.3.1 and cmake 3.18.2)

ayoubSoussi, 2020-12-08:
It works fine also in Ubuntu 18.04 but with the combination of commands that I described in the previous comment.

salkinium, 2021-03-29:
I'm closing this issue since we cannot reproduce this bug and you seem to have found a workaround.

closed 2021-03-29 by salkinium