modm-io/modm #517: [Question] I2C module in RPI

ayoubSoussi, 2020-12-12:
Hi,

I am trying to add a new I2C module to the Raspberry Pi Platform. It's basically a wrapper of the library WiringPi.
In the file I2C_master.cpp, I'm a stuck in the implementation of a function that executes a transaction. I couldn't understand well how I2C_transaction s work exactly (for example what to do when the next transaction is a Restart).

This is the function that executes a transaction (called in the startTransaction function)
[image]

I added just Write operation (as a first step).
To sum up, my question is 

- What should I do when the nextOperation is a Restart ?

Thanks.

ayoubSoussi, 2020-12-12:
Apparently WiringPiI2C contains only basic functions for reading and writing, perhaps it would be better to interact with the i2c kernel driver directly.

salkinium, 2020-12-12:
I mean you can try to do a write-stop-start-read sequence to get you started with something, but it's functionally different than a write-restart-read sequence and some devices will not work correctly. But it may work for your devices for now and I'll be happy to merge this with a big disclaimer about restart conditions that someone else can then fix.

ayoubSoussi, 2020-12-12:
I want to use for an I2C SSD1306 LCD, and after looking at the SSD1306 driver I see that it uses restart so I'm not sure if it will work without modifying the SSD1306 driver.

salkinium, 2020-12-12:
You can use the RPI GPIO drivers in our Software I2C Bitbang driver, which is feature complete. It may not be very efficient and it'll be blocking, but it should work fine.

salkinium, 2020-12-12:
Although I'm not sure how to set OpenDrain mode with WiringPi, but @henrikssn used it with our I2C and SPI Bitbang drivers I believe?
https://github.com/modm-io/modm/blob/develop/src/modm/platform/gpio/rpi/pin.hpp

henrikssn, 2020-12-12:
I only tried the SPI bitbang interface with RPi, it worked fine at a modest 50kHz clock.

closed 2021-09-29 by salkinium