modm-io/modm #539: Debug build with GCC10.2.0

XDjackieXD, 2021-01-15:
When building a debug build with GCC 10.2.0, the following assembler error is emitted:
```
/tmp/cc7OQ88c.s: Assembler messages:
/tmp/cc7OQ88c.s:7806: Error: Thumb-2 MUL must not set flags -- `muls.n r2,r2,ip'
/tmp/cc7OQ88c.s:7808: Error: cannot honor width suffix -- `subs.n r4,r4,ip'
/tmp/cc7OQ88c.s:8015: Error: cannot honor width suffix -- `subs.n lr,lr,r1'
/tmp/cc7OQ88c.s:8016: Error: cannot honor width suffix -- `subs.n lr,lr,r4'
```
Release builds do not suffer from the same problem.
It seems to be related to delay_impl.hpp

rleh, 2021-01-15:
Which target are you compiling for? (STM32, SAM, hosted/arm64/linux, ...)

Can you provide more information how to reproduce the bug?

salkinium, 2021-01-15:
Specifically: are you compiling for ARMv6-M (Cortex-M0/M0+)? Perhaps the assembler is missing some instruction  transformations in debug mode? Or we messed up and these instructions are indeed unavailable, but were silently fixed before.

XDjackieXD, 2021-01-15:
Ah sorry I forgot to write that...
I'm building for the STM32F303.
It was working with earlier GCC versions (I don't know though, which exact version it stopped working)

chris-durand, 2021-01-15:
I tried to reproduce it by building Nucleo F303 examples in debug mode (with `lbuild build; scons -j8 profile=debug`) but it builds fine on my machine.
Could you try to compile the Nucleo F303 examples on your system? What does `arm-none-eabi-g++ --version` print on your machine? What project are you building, a custom one or a modm example?

salkinium, 2021-01-15:
Are you using the precompiled ARM GCC by ARM? or the one bundled by your OS's package manager? The latter are typically built with slightly different patches/flags.

rleh, 2021-01-15:
I cannot reproduce the problem on Fedora 32 (`arm-none-eabi-g++ (Fedora 10.2.0-2.fc32) 10.2.0`) from fedora packages), nor on Ubuntu 20.04 with the GCC distributed by ARM.

```
cd examples/nucleo_f303re/blink/
lbuild build
scons -j8 profile=debug
```

XDjackieXD, 2021-01-15:
The blink example compiles fine.
my project however triggers it reliably with `cmake .. -DCMAKE_BUILD_TYPE=Debug && make` (without the debug build type, everything works fine).

this is my project.xml:
```
<library>
	<repositories>
		<repository>
			<path>../modm-library/repo.lb</path>
		</repository>
	</repositories>

	<extends>modm:tuwr_generic_f303cb</extends>

	<options>
		<option name=":build:project.name">e12-precharge-firmware</option>
		<option name="modm:platform:cortex-m:main_stack_size">10240</option>
		<option name=":build:build.path">cmake-build-cmdline</option>
	</options>

	<modules>
		<module>modm:platform:gpio</module>
		<module>modm:platform:spi:1</module>
		<module>modm:platform:spi:2</module>
		<module>modm:platform:can</module>
		<module>modm:platform:timer:1</module>
		<module>modm:platform:timer:2</module>
		<module>modm:driver:ltc6810</module>
		<module>modm:platform:uart:2</module>
		<module>modm:platform:fault</module>
		<module>modm:crashcatcher</module>
		<module>modm:processing:timer</module>
		<module>:build:cmake</module>
		<module>:build:scons</module>
	</modules>
</library>
```

my gcc/g++ version is
```
arm-none-eabi-g++ (Arch Repository) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
(directly from the arch repo)

I'll try to strip my code down to the bare minimum to trigger this during the next days

salkinium, 2021-01-15:
> arm-none-eabi-g++ (Arch Repository) 10.2.0

Can you try with the official ARM GCC? That's the recommended compiler, as we've had such "weird" issues before: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

rleh, 2021-01-15:
> `cmake .. -DCMAKE_BUILD_TYPE=Debug && make`

I would also suspect CMake or our CMake build generator.
But I tested it on Ubuntu with the nucleo_f303re/blink example, but I couldn't reproduce your problems there either.

XDjackieXD, 2021-01-15:
> But I tested it on Ubuntu with the nucleo_f303re/blink example, but I couldn't reproduce your problems there either.

I can't reproduce this issue with the blink examples either.
As I already said I will try to pinpoint this to the minimum viable project that still triggers this bug and upload it for you to test.

> Can you try with the official ARM GCC?

I will try this either tomorrow or on Sunday and report back

XDjackieXD, 2021-01-15:
I've now managed to slim down my project to the minimum (a few lines of code, a minimum project.xml and built by scons) necessary to trigger the bug on a Arch system (see attached project.xml and main.cpp; it expects modm cloned as "modm-library" into the parent directory; I've used current develop branch for testing).
I'll try with the official arm-gcc next.
[minimum-viable-trigger-539.zip](https://github.com/modm-io/modm/files/5823370/minimum-viable-trigger-539.zip)

XDjackieXD, 2021-01-15:
Same problem with official ARM-GCC:

```
> arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
> scons -j8 profile=debug
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Compiling C++·· build/minimum-viable-bug-trigger/debug/main.o
Compiling C···· build/minimum-viable-bug-trigger/debug/modm/ext/gcc/cabi.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/ext/gcc/cxxabi.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/ext/gcc/new_delete.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/architecture/driver/atomic/flag.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/board/board.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/io/iostream.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/io/iostream_printf.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/math/utils/bit_operation.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/math/utils/pc/operator.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/clock/rcc.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/clock/systick_timer.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/core/assert.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/core/delay.o
Compiling C···· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/core/no_heap.o
Assembling····· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/core/reset_handler.o
Compiling C···· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/core/startup.o
Compiling C···· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/core/startup_platform.o
Compiling C···· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/core/vectors.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/gpio/enable.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/platform/uart/uart_2.o
Compiling C++·· build/minimum-viable-bug-trigger/debug/modm/src/modm/utils/dummy.o
/tmp/ccFetWbH.s: Assembler messages:
/tmp/ccFetWbH.s:925: Error: Thumb-2 MUL must not set flags -- `muls.n r2,r2,ip'
/tmp/ccFetWbH.s:927: Error: cannot honor width suffix -- `subs.n r1,r1,ip'
scons: *** [build/minimum-viable-bug-trigger/debug/main.o] Error 1
scons: building terminated because of errors.
```

chris-durand, 2021-01-16:
I can reproduce it with Ubuntu and the official ARM GCC:

`scons -j8 profile=debug`:
```
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Compiling C++·· build/minimum-viable-trigger-539/debug/main.o
Compiling C···· build/minimum-viable-trigger-539/debug/modm/ext/gcc/cabi.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/ext/gcc/cxxabi.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/ext/gcc/new_delete.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/architecture/driver/atomic/flag.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/board/board.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/io/iostream.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/io/iostream_printf.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/math/utils/bit_operation.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/math/utils/pc/operator.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/clock/rcc.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/clock/systick_timer.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/core/assert.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/core/delay.o
Compiling C···· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/core/no_heap.o
Assembling····· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/core/reset_handler.o
Compiling C···· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/core/startup.o
Compiling C···· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/core/startup_platform.o
Compiling C···· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/core/vectors.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/gpio/enable.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/platform/uart/uart_2.o
Compiling C++·· build/minimum-viable-trigger-539/debug/modm/src/modm/utils/dummy.o
/tmp/cchkt8mu.s: Assembler messages:
/tmp/cchkt8mu.s:925: Error: Thumb-2 MUL must not set flags -- `muls.n r2,r2,ip'
/tmp/cchkt8mu.s:927: Error: cannot honor width suffix -- `subs.n r1,r1,ip'
scons: *** [build/minimum-viable-trigger-539/debug/main.o] Error 1
scons: building terminated because of errors.
```

chris-durand, 2021-01-16:
Compiles in debug mode:
```
#include <modm/board.hpp>

using namespace Board;

int main()
{
	Board::initialize();

	GpioB1::set();
	modm::delay_ns(1);
	GpioB1::set();
	modm::delay_ns(1);

        uint32_t counter(0);
	while (true)
	{
    	    MODM_LOG_INFO << counter++;
	}

	return 0;
}
```

Triggers the error in debug mode:
```
#include <modm/board.hpp>

using namespace Board;

int main()
{
	Board::initialize();

	GpioB1::set();
	modm::delay_ns(1);
	GpioB1::set();
	modm::delay_ns(1);

	while (true)
	{
    	    MODM_LOG_INFO << "test";
	}

	return 0;
}
```

Looks like a compiler bug ...

salkinium, 2021-01-16:
No, the compiler is correct, I believe the default register allocation strategy changed. This also explains why some code changes trigger this and others don't.

In my example, the compiler is allocating the high register `lr` (r14) to be used:
```
muls.n r2,r2,lr
subs.n r1,r1,lr
```

However the 16-bit Thumb2 encoding only allows r0-r7 to be used:
[image]

This means the asm register constraints inside modm is wrong, it should only allow the lower registers to be substituted. [By using the `l` constraint, it works correctly](https://gcc.gnu.org/onlinedocs/gcc/Machine-Constraints.html#Machine-Constraints).

salkinium, 2021-01-16:
It's really quite outrageous how much you can do wrong with inline asm. I wonder is there a additional flags we can use, like how printf has a format checker? Probably not. Grrrr.

PR #540 (2021-01-16): [delay] Fix register constraints for subs.n instruction
Fixes #539.

cc @chris-durand @rleh @XDjackieXD
closed 2021-01-16 by rleh