modm-io/modm #549: Undefined reference __exidx_end after switching to GCC 10.x

mikewolfram, 2021-02-02:
Hi all,

maybe somebody has a hint to resolve this issue. After switching from GCC 9.2 to 10.2 I get this error during linking:
```
unwind-arm.c:(.text.get_eit_entry+0x94): undefined reference to `__exidx_start'
/opt/xpack-arm-none-eabi-gcc-10.2.1-1.1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: unwind-arm.c:(.text.get_eit_entry+0x98): undefined reference to `__exidx_end'
```

I build with exceptions and unwind table disabled. Using the ARM GCC from xpack on macOS.

salkinium, 2021-02-02:
> I build with exceptions and unwind table disabled.

You definitely need the unwind tables with exceptions enabled.
modm should [automatically include the unwind tables](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/linker.macros#L220-L248) in the linkerscript if you set the [`:stdc++:exceptions` option](https://modm.io/reference/module/modm-stdc%2B%2B/#exceptions) to `true`.

salkinium, 2021-02-02:
Oh I see, you mean you build with both exceptions disabled *and* unwind tables disabled.

In that case it may be that the toolchain profile includes the unwind code even with exceptions disabled. Does this issue occur with the official ARM toolchain too? Otherwise we need to provide the symbols in the linkerscript even if we discard the section.

mikewolfram, 2021-02-02:
Well, this is my workaround for the moment, but I wasn't expecting that I need exceptions enabled. Is there a specific reason for that?

mikewolfram, 2021-02-02:
Too slow... :-)

Yes, both disabled, exceptions and unwind table. So I'll go and try the ARM toolchain.

salkinium, 2021-02-02:
Homebrew for the win:
```
brew tap osx-cross/arm
brew install arm-gcc-bin
```

mikewolfram, 2021-02-02:
I'm using MacPorts, but their latest version is 9.2. That's why I've chosen xpack, since I also use Eclipse.

The ARM GCC 10.2 resolves the issue. Could this be considered a bug, that should be told to the xpack team?

closed 2021-02-02 by mikewolfram

salkinium, 2021-02-02:
I guess? We've had lots of weird issues with the non-official ARM compilers, it's usually very difficult to get all the flags and options just right, because ARM didn't upstream the build script they use.

Fun fact: I spend a couple weeks porting the build script into a standalone Homebrew tap, unfortunately since I was the only macOS user there, so they didn't publish the tap after I left… and now I've already forgotten what I did to get it to work in the first place…

mikewolfram, 2021-02-02:
I know how that feels. Did the same in the company, preparing scripts etc. for easy use. Nobody cared and I'll be gone soon...