modm-io/modm #550: Separate linker section for startup code

mikewolfram, 2021-02-03:
Ciao,

I'd like to propose moving the functions required for startup to a separate code section. I'm working on a boot loader, which will run completely from RAM to support updating itself as well.

As of now I identified these functions to be in a "startup" section:
* Reset_Handler
* table_call
* table_copy
* table_zero
* __modm_initialize_platform
* __modm_startup

(Did I miss one?)

Doing it this way it would be easy to create an application that can run from RAM (after adjusting the linker script).

What is you opinion?

salkinium, 2021-02-03:
You will need to use a custom linkerscript, by copying the generated one and replacing `FLASH` with `RAM` everywhere. You can use the [`:platform:cortex-m:linkerscript.override` PathOption](https://modm.io/reference/module/modm-platform-cortex-m/#linkerscriptoverride) to point to a custom linkerscript.

However, you don't need to place the entire application in RAM, only the code that erases/programs the Flash. The `Flash::erase()` and `Flash::program()` functions are executed from RAM via `modm_ramcode` for that purpose. You can even update your own bootloader by first copying the bootloader image to another section in Flash, then calling this ramcode function ([from here](https://gist.github.com/salkinium/fb0e4aef572d909c9ded1594dd081f1d)):

```cpp
modm_ramcode
void ram_apply(uint8_t pages, uint32_t length)
{
	modm::atomic::Lock l;
	for (auto page{0u}; page <= pages; page++)
		Flash::erase(page);

	for (uintptr_t ptr{0}; ptr < length; ptr += sizeof(Flash::MaxWordType))
		Flash::program(Flash::OriginAddr + ptr, *(Flash::MaxWordType*)(BUFFER_START + ptr));

	NVIC_SystemReset();
}
```

You don't even need to have a bootloader for this to work, this can be part of the application itself, which is much, much simpler. However you can also brick your device by not having a *separate* bootloader…

salkinium, 2021-02-03:
It would be useful to place the bootloader first into Flash, and then delegate to the Application. For this to work, there's [option `:platform:cortex-m:flash_offset`](https://modm.io/reference/module/modm-platform-cortex-m/#linkerscriptflash_offset) That moves the entire FLASH section forward by whatever offset you specify. That way you don't need a custom linkerscript for either the bootloader (no offset) nor the application (offset compatible with a Flash page).

mikewolfram, 2021-02-03:
Thanks for the explanation. I already have a modified linker script and changed the mentioned functions to be placed in a separate section. So this part is working. I wanted to ask if we could place them generally into their own section to avoid having my own copy of the files.

I have limited space in flash, so I cannot download the boot loader to a different location in flash and copy it later.

salkinium, 2021-02-03:
I'm not sure selectively moving functions into RAM is a good idea, since you also need to move every function called from `__modm_startup` into RAM. At that point you will collide with the `.hardware_init` and `.init` constructor sections and everything they call. That's a lot of functions.

Do you have enough RAM to fit the entire bootloader in it? Then you could just copy the whole bootloader image to RAM and execute it there.

mikewolfram, 2021-02-03:
You mean use a two step boot loader? First step is to copy the second step from flash to RAM and then jump into it? Could work, the boot loader itself is less than 20kB and runs completely from RAM anyway.

The `.init` section can be in flash or RAM (as now I have it copied to RAM), for the `.hardware_init` the same, I keep/run it from flash right now.

No matter how the actual application is structured at the end, I still believe it would make sense to move the mentioned function to their own section. The default linker script could add them to the `.text` section.

salkinium, 2021-02-03:
Still not convinced 🧐 The `.init` sections only contains pointers to functions and you don't know exactly which functions. It could also be compiler/linker generated veneers for jumping larger distances. So you don't have any guarantees unless you move the whole `.text` section. At that point it's easier to allocate it all to RAM, and let something else copy it into place.

mikewolfram, 2021-02-03:
This is what I do, copy the whole `.text` to RAM. That's why I need the startup functions stay in flash.

The way I do it right now is pretty easy, I don't need an extra stub to copy the image to RAM first.

salkinium, 2021-02-03:
Ahhhhhhh, I see of course. That doesn't require two separate images only for one bootloader then.

We should add an EnumerationOption that specifies the level of RAMness, like "Flash-only no copy" (as it is now), "Flash to RAM selfcopy" and "RAM-only external copy". That way we can explicitly change the startup and linker script, rather than inferring this via section placement alone?

mikewolfram, 2021-02-03:
If this were with live images you would see me confused? :-)

Maybe you could explain a bit more?

salkinium, 2021-02-03:
I think it may "just" be this?

1. Change the linkerscript to move all the `>FLASH`  to `>RAM AT >FLASH`. Perhaps you need to move sections around to avoid some circular math issues (linker can't just back).
2. Empty the internal copy table, since that's already in RAM. The zero table and the external tables need to stay.
3. Insert the copy call into the [`reset_handler.sx` file after `__modm_initialize_platform`, but before `__modm_startup`](https://github.com/modm-io/modm/blob/develop/src/modm/platform/core/cortex/reset_handler.sx#L22) that copies the whole image to RAM incl. text and data.
4. Relocate the vector table into RAM. There's an lbuild option for that, but you must overwrite it here.

Then from the jump to `__modm_startup` onwards you should only execute in RAM I think?

salkinium, 2021-02-03:
And I forgot the most important thing: The ResetHandler and copy code must remain in flash, thus they must be placed there by the linkerscript or via a `modm_section`.

mikewolfram, 2021-02-03:
Well, actually I wanted to know more about the enumeration thing you mentioned above.

My code is working perfectly fine using the default files of modm (patched to move a few functions to a `.startup` section) and a custom linker script. The functions in the first comment stay in flash, everything else is copied to RAM in `__modm_startup`. 

I don't want to make it too complicated. That's why the idea of the `.startup` section.

I'm now reading the LD docs. Since I use a custom linker script anyway I'll to place code by filename to that section.

salkinium, 2021-02-03:
I see three scenarios that can be enumerated in an option:

1. The bootloader boots and executed from Flash like a normal application. It can update itself by storing a copy of the new Bootloader image in left over Flash and then jumping to a ramcode function that copies the image over atomically then reboots.
2. The bootloader boots from Flash but executed from RAM. This allows to overwrite the bootloader image on-the-fly without first storing a copy.
3. The bootloader boots from RAM and executes from RAM. This allows for any bootloader to be transmitted over a network and copied in to RAM by the application, allowing to update the whole Flash with just the application.

mikewolfram, 2021-02-03:
Well, got the the linker script changed to move the functions mentioned to its own section without adding a `modm_section(".startup") to every function. So I don't really need it inside modm anymore while it still might be a good idea to have them in a separate section.

For our project we don't have enough space in flash to download a copy of the boot loader and then move in flash. So I need to run from RAM the whole boot loader. Your 3rd option also sounds like an idea, download the boot loader (or better call it flash tool?) to the board and execute it, just for updating. But this isn't reasonable for us. We have a few hundreds PCB installed and more to come, all connected via CAN. Whenever we order new I want the boot loader preinstalled and then simply do a mass update. Unfortunately the built-in boot loader of ST doesn't work as expected. There's no way to set the CAN ID.

Since the modified linker script works for me, we could close this one. If you want the separate section, I could open a PR to submit changes for a dedicated `startup` section.

mikewolfram, 2021-02-03:
Looking into this I got another question. Isn't there a place to contribute some how-to guides? What we do might not be usual, but could be of interest for others.

salkinium, 2021-02-04:
Yes, here: https://github.com/modm-io/modm/tree/develop/docs/src/guide

Let's leave this issue open, since I'm also currently writing an In-Application Programmer over CAN (scenario 1 though).

mikewolfram, 2021-02-04:
Just in case somebody wants to know how to do it:
```
	.startup : ALIGN(4)
	{
		*reset_handler.o(.text .text.*)
		*startup.o(.text .text.*)
		*startup_platform.o(.text .text.*)
		*(.startup)
		. = ALIGN(4);
	} >FLASH
```
I didn't had to add the files as exclusion in the generic text section.

The linker will also create two veneers in the section, one for `__modm_initialize_memory` and then for `main`.

salkinium, 2021-05-08:
I solved the CAN IAP without linkerscript changes, therefore I'm closing this.

closed 2021-05-08 by salkinium