modm-io/modm #551: [Question] Generate and use libmodm.a+headers only? (no linking to main)

mfp20, 2021-02-08:
**Is there a proper way to build the static library, and a folder for headers, only?**

Without a main function, the library gets generated and all the modm generated source code (hpp+cpp) is in a subfolder. It stops at 'undefined reference to main' when linking to build the firmware.

**What about using the library then?**

If I cd in avr logger example and try to manually build it

`avr-g++ -o logger.elf -Imodm/ext/gcc/libstdc++/include -Imodm/src -L../../../build/avr/logger/release/modm -lmodm -std=c++20 main.cpp`

results in a huge scroll of errors filling up my terminal history. Looks like all the PORTx PINy definitions missing and more.
And scons doesn't help much in figuring out the proper build flags because those are in variables; if I build with scons --debug presub all I get are rows similar to this one

`$CXX -o $TARGET -c $CXXFLAGS $CCFLAGS $_CCCOMCOM $SOURCES`

and I don't know how to expand those variables. Looks like scons doesn't give a flag for that. Do I have to disable the generation of SConstruct and mod it?

salkinium, 2021-02-08:
You can use `scons verbose=1` to output the actual calls to the compiler/linker.

For `libmodm.a`: This is already implemented internally as a static lib + headers.
We first build a static library called `libmodm.a` and then link that to your application with the include paths. For the `modm/example/avr/logger`, you can find the static library in `modm/build/avr/logger/release/modm/libmodm.a`.

```
Create Library· modm/build/avr/logger/release/modm/libmodm.a
Indexing······· modm/build/avr/logger/release/modm/libmodm.a
Linking········ modm/build/avr/logger/release/logger.elf
Memory usage··· modm/build/avr/logger/release/logger.elf
```

The [generated `modm/SConscript` implements the `libmodm.a` static library compilation](https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/scons/resources/SConscript.in#L159), sets all the relevant global compiler/linker flags for avr-g++ and adds the generated static library to the global `LIBS` variable.

If you only want to generate the `libmodm.a` (for example to archive it in a repository), you can modify your SConstruct like this, and call `scons library`.

```diff
- env.SConscript(dirs=generated_paths, exports="env")
+ libraries = env.SConscript(dirs=generated_paths, exports="env")
+ env.Alias("library", libraries[0])
```

You can also delete everything below those lines, if you *only* want to generate the static lib.

salkinium, 2021-02-08:
If this solves your issue, I can add `scons library` or similiar to the default SCons arguments, so that you don't even need to modify your SConstruct files. This could be helpful for automating libmodm.a generation in a CI or similiar.

salkinium, 2021-02-08:
I also just found a bug, where SCons adds `-L{project_source_dir}` to the final link output, which isn't formatted correctly, but is apparently not an issue for compilation 😳.

mfp20, 2021-02-08:
> If this solves your issue, I can add `scons library` or similiar to the default SCons arguments, so that you don't even need to modify your SConstruct files. This could be helpful for automating libmodm.a generation in a CI or similiar.

**That would be awesome!** I've been searching for it a couple of days but I couldn't make it. If I use StaticLibrary instead of BuildTarget() or Return() ... I simply get nothing.
Ideally, it should generate the libmodm.a file and an "include/" subfolder reproducing the tree of hpp files.

salkinium, 2021-02-08:
The include path is defined by this section in the `modm/SConscript` (relative to that file):

```py
env.AppendUnique(CPPPATH=[
    abspath(r"ext"),
    abspath(r"ext/gcc/libstdc++/include"),
    abspath(r"src"),
])
```

So you can basically copy these folders whereever you want, and then set the `-Ipath` flags accordingly. You can remove all source files from those paths too (perhaps even with a `.gitignore`).

For you that would mean:

1. Create an empty project without `main.cpp`, only with `project.xml` with all the settings/modules you want (build path is defaulted to `.` outside of the modm example folder).
2. `lbuild build` and copy the generated `modm/ext`, `modm/ext/gcc/libstdc++/include`, `modm/src` folders wherever you need them. (Remove or .gitignore the source files)
4. `scons libraries` and copy the generated `libmodm.a` whereever you need it.
5. Copy the compile flags out of `scons verbose=1` or from the `modm/SConscript` into your own build system.
6. Profit? I guess…

I'll fix the format bug and add `scons libraries` as a default target.

salkinium, 2021-02-08:
Please remember to use `-Wl,--whole-archive -Wl,--start-group libmodm.a -Wl,--end-group -Wl,--no-whole-archive` for linking, otherwise weak functions inside libmodm may not be linked correctly.

PR #552 (2021-02-08): Add scons library target to generate only static library
This can be useful to extract libmodm.a + headers into a custom build system without having to add a `main.cpp`.

Closes #551.
cc @mfp20
closed 2021-02-08 by salkinium