modm-io/modm #554: STM32F407 Discovery Board - USB

lukh, 2021-02-10:
Hi ! 
I was trying to run the sample project (examples/generic/usb) with the "modm:disco-f407vg" target.
The board is the stm32f4-discovery obviously, rev MB997C.

I am really annoyed to come and seek help, but this is the first time I work with USB.
I built a basic example code, build with STM32CubeMX, seen as a HID device, it does work (after adding the pullup 1.5k on one of the Dataline), so I doubt the hw is not working properly...

This example should work out of the box right ? No need to implement weak callbacks, or fill descriptors somewhere in the tinyusb sub folder ? (at least, to make this example running)

It is seen as a USB device, but windows tell me a Code43 error, could not get the usb descriptor.

What did I miss ?
Thanks for your help !
Vivien

rleh, 2021-02-10:
> This example should work out of the box right?

Yes.

I just tested the  `examples/generic/usb` example on `modm:disco-f407vg` target and it works fine without any hardware modification on my (linux) computer:

```bash
$ lsusb
# ...
Bus 003 Device 006: ID cafe:4005 TinyUSB TinyUSB Device

$ dmesg
# ...
[118546.698426] usb 3-1: New USB device found, idVendor=cafe, idProduct=4005, bcdDevice= 1.00
[118546.698428] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[118546.698429] usb 3-1: Product: TinyUSB Device
[118546.698430] usb 3-1: Manufacturer: TinyUSB
[118546.698431] usb 3-1: SerialNumber: 004C003B3434511034383735C000FCC0
[118546.699327] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
[118546.699659] usb-storage 3-1:1.2: USB Mass Storage device detected
[118546.700468] scsi host7: usb-storage 3-1:1.2
[118547.760705] scsi host7: scsi scan: INQUIRY result too short (5), using 36
[118547.760712] scsi 7:0:0:0: Direct-Access     TinyUSB  Mass Storage     1.0  PQ: 0 ANSI: 2
[118547.761026] sd 7:0:0:0: Attached scsi generic sg5 type 0
[118547.761329] sd 7:0:0:0: [sde] 16 512-byte logical blocks: (8.19 kB/8.00 KiB)
[118547.761529] sd 7:0:0:0: [sde] Write Protect is off
[118547.761532] sd 7:0:0:0: [sde] Mode Sense: 03 00 00 00
[118547.761693] sd 7:0:0:0: [sde] No Caching mode page found
[118547.761695] sd 7:0:0:0: [sde] Assuming drive cache: write through
[118547.771339]  sde:
[118547.772688] sd 7:0:0:0: [sde] Attached SCSI removable disk
```

I currently do not have access to a Windows-PC, maybe someone else can help with this?
Most modm developers do not use Windows, so it can happen that bugs are overlooked that only occur with Windows.

salkinium, 2021-02-10:
A couple of things I can think of:

- You can update the TinyUSB submodule in `ext/hatach/tinyusb`. I did that just now, but  the generic/usb example works for me on macOS too both before and after update.
- You can try the official TinyUSB example for the F407. However, modm's setup should be identical.
- Perhaps Windows expects a different Vendor/Product ID? What was used by CubeMX?

chris-durand, 2021-02-10:
Windows saves the mappings for Vendor + Product Ids to device driver somewhere in the registry. If you connected a different device with the same ids it might fail, see https://stackoverflow.com/questions/41644081/windows-10-clean-out-usb-enumeration-driver-cruft.

lukh, 2021-02-11:
2760.887534] usb 1-4: new full-speed USB device number 64 using xhci_hcd
[ 2761.030120] usb 1-4: New USB device found, idVendor=cafe, idProduct=4005, bcdDevice= 1.00
[ 2761.030128] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2761.030133] usb 1-4: Product: TinyUSB Device
[ 2761.030137] usb 1-4: Manufacturer: TinyUSB
[ 2761.030140] usb 1-4: SerialNumber: 002C00343233470238373334C000FCC0
[ 2761.033310] cdc_acm 1-4:1.0: ttyACM0: USB ACM device
[ 2761.034171] usb-storage 1-4:1.2: USB Mass Storage device detected
[ 2761.034644] scsi host0: usb-storage 1-4:1.2
[ 2762.168167] usb 1-4: reset full-speed USB device number 64 using xhci_hcd
[ 2762.290742] usb 1-4: device descriptor read/64, error -71

lukh, 2021-02-11:
Ok....
I was able to make tinyusb example work, on the unmodified stm32f407disco, and another board as well (stm32f072disco)...
(cdc/msc example), I can see (on linux a new /dev/ttyACM0 and a media storage)...
So examples in tinyusb examples work...

Next step, import tinyusb in a modm project by hand I guess ?

salkinium, 2021-02-11:
No, hang on, that's too much work. The TinyUSB core code is copied unmodified by lbuild into `modm/ext/tinyusb`, so you only need to check the differences, I suspect the descriptors to be the issue:

The things modm adds or does different are the following (all to be found in the generated `modm/ext/tinyusb` folder):
- USB descriptors in [`tusb_descriptors.c`](https://github.com/modm-io/modm/blob/develop/ext/hathach/tusb_descriptors.c.in). Since the error is about `device descriptor read/64`, I suspect this is the issue.
- IRQ generation and UID serial generation in [`tusb_port.cpp`](https://github.com/modm-io/modm/blob/develop/ext/hathach/tusb_port.cpp.in). However, since the device is getting found, probably not the issue.
- Board Setup: Specifically the [`Board::initializeUsb()`](https://github.com/modm-io/modm/blob/develop/src/modm/board/disco_f407vg/board.hpp#L224-L235) from here particularly VBUS setup is difficult. However, if you get this wrong the device doesn't even enumerate at all, so probably not the issue here.

salkinium, 2021-02-11:
Comparing the descriptors between modm and [the TinyUSB cdc_msc example](https://github.com/hathach/tinyusb/blob/master/examples/device/cdc_msc/src/usb_descriptors.c):

- modm uses a different `USB_PID`, because when using two CDCs it sets the MSC bit. I think this is a bug in TinyUSB, I need to open an issue.
- `ITF_NUM_*` is identical.
- `EPNUM_*` is identical.
- `desc_fs_configuration` is identical.
- modm doesn't generate `desc_hs_configuration` for this device, since the hw can only do FS anyways. Perhaps I misunderstood something there?
- `string_desc_arr` is almost identical. The serial number is NULL, since it's replaced in code.
- `tud_descriptor_string_cb` is adapted to call the chip serial, but otherwise identical.

salkinium, 2021-02-11:
Unfortunately the STM32F4 Discovery does not have a serial output, so you don't see the TinyUSB assertions on the serial port. But perhaps if you do `scons program profile=debug && scons debug profile=debug` and then set a breakpoint on `b modm_assert_report` then `rerun`, you will halt on an assertiona and can print the assertion information: `p cinfo`.

lukh, 2021-02-11:
Thanks for the help!
So, with modm code, the unmodified board stm32f407 is not found anymore

lukh, 2021-02-11:
I'd suspect HW/Peripheral configuration ?

salkinium, 2021-02-11:
[NVIC and connect](https://github.com/modm-io/modm/blob/develop/src/modm/platform/usb/stm32/usb.hpp.in) are all done by the peripheral.
[All GPIO ports are enabled](https://github.com/modm-io/modm/blob/develop/src/modm/platform/gpio/stm32/enable.cpp.in) by default.

lukh, 2021-02-11:
Ok, thank you !
Since the device doesn't enumerate at all, 
"Board Setup: Specifically the Board::initializeUsb() from here particularly VBUS setup is difficult. However, if you get this wrong the device doesn't even enumerate at all, so probably not the issue here." What you said earlier make sense right ?

lukh, 2021-02-11:
ok found the solution

lukh, 2021-02-11:
`Id::configure(Gpio::InputType::PullUp);`

lukh, 2021-02-11:
in usb_fs.hpp, after line 45

salkinium, 2021-02-11:
Oh no! You're absolutely right, the TinyUSB BSP sets this as PullUp too…
https://github.com/hathach/tinyusb/blob/master/hw/bsp/stm32f4/family.c#L114-L120

Do you want to open a PR for this fix?

lukh, 2021-02-11:
hehe, yes, I was comparing low level init in both codes, and then, ...

Yes, i'll do  it !

Thank a lot for the help by the way, would 'nt have found it otherwise (at least, not in one day ^^)

closed 2021-02-11 by lukh