modm-io/modm #563: Should protothreads cause a significant amount of overhead?

ceremcem, 2021-02-19:
Hi, 

For this code: 

```cpp
#include "board.hpp"
#include <modm/processing/protothread.hpp>
#include <modm/processing/timer.hpp>

using namespace Board;

class BlinkingLight : public modm::pt::Protothread
{
public:
    bool
    run()
    {
        PT_BEGIN();

		LedGreen::set();

        while (true)
        {
            timeout.restart(100ms);
            LedGreen::set();
            PT_WAIT_UNTIL(timeout.isExpired());

            timeout.restart(200ms);
            LedGreen::reset();
            PT_WAIT_UNTIL(timeout.isExpired());
        }

        PT_END();
    }

private:
    modm::ShortTimeout timeout;
};

int
main()
{
	Board::initialize();
	BlinkingLight light;

	while (true)
	{
		ClockPin::set();
		modm::delay(0.2us);

		ClockPin::reset();
		modm::delay(0.2us);
		//light.run();
	}

	return 0;
}
```

When this code runs on STM32F103C8T6, I measure a 2MHz square wave on `ClockPin`. When I uncomment `//light.run()` line, the green led on the BluePill blinks as expected, however, the frequency of `ClockPin` drops down to 444kHz.

What is wrong with that approach?

rleh, 2021-02-19:
What core frequency is your STM32F103 running?
Are you compiling with release or debug profile (`scons profile=debug ...`)? (default: _release_)

salkinium, 2021-02-19:
So the F103 has a max CPU frequency of 72MHz or ~14ns per cycle. A delay of 0.2µs = 200ns = ~14 cycles, which makes is very difficult to do anything really. In this context I'm actually impressed the Protothread only slows down the ClockPin 4x. Also keep in mind that the delay nanoseconds also has an overhead and is very accurate at these super short delays.

> What is wrong with that approach?

Bit Banging in software becomes difficult on the STM32 beyond the MHz scale. You really need to use some hardware peripheral for this or bit bang the protocol via Timer triggered DMA to GPIO.

ceremcem, 2021-02-19:
I'm compiling with `scons program` command, so it's compiled with release profile. 

> What core frequency is your STM32F103 running?

It's actually the modified version of https://github.com/modm-io/modm/tree/develop/examples/stm32f103c8t6_blue_pill/blink example, so all settings is the same as that example. Which means, `72_MHz` I guess: 

```cpp
/// STM32F103 running at 72MHz generated from the external 8MHz crystal
// Dummy clock for devices
struct SystemClock {
	static constexpr uint32_t Frequency = 72_MHz;
	static constexpr uint32_t Ahb = Frequency;
	static constexpr uint32_t Apb1 = Frequency / 2;
	static constexpr uint32_t Apb2 = Frequency;
```

salkinium, 2021-02-19:
Did my comment answer your question?

closed 2021-02-19 by ceremcem

ceremcem, 2021-02-19:
It shed a light on the case, yes. I was experimenting some things since your answer. The cycle count you provided actually tells everything. 

For example, the following code also introduces the same amount of overhead as that Protothread:

```cpp
	while (true)
	{
            for(int i=0; i < 9; i++){
                ClockPin::set();
	    }
	    modm::delay(0.2us);
	    for (int i=0; i < 9; i++){
                ClockPin::reset();
    	    }
            modm::delay(0.2us);
	}
```

So, if 18 simple loops create the same amount of overhead, protothreads are not that heavyweight after all

salkinium, 2021-02-19:
> So, if 18 simple loops create the same amount of overhead, protothreads are not that heavyweight after all

That's good to hear!

The Protothread itself is probably even faster, the most time is likely spent on calling the `*Timeout::isExpired()` method, which calls `modm::Clock::now()`. I spend a lot of effort making that timekeeping code very fast too, so I'm very happy that it payed off!

Here's a small thread in case you're interested in the details:
https://twitter.com/modm_embedded/status/1270029826641051649