modm-io/modm #567: make freeRtos thread wrapper more dynamic

kikass13, 2021-02-22:
At the moment, the wrapper implementation of modm::Thread inside the freeRtos module is useless, because the `modm::rtos::Thread::Thread` creates the rtos task directly inside the constructor

```cpp
modm::rtos::Thread::Thread(uint32_t priority,
		uint16_t stackDepth,
		const char* name)
{
	xTaskCreate(
			&wrapper,
			name ? name : "anon",
			(stackDepth / 4) + 1,
			this,
			priority,
			&this->handle);
}
```

This makes no real sense to me, the constructor should be empty. Because Task can be created dynamically (will malloc anyways at the point of construction) one should be able to declare Threads without running them - and run them at a later stage in the programm (interrupt controlled for example) respectively

salkinium, 2021-02-22:
modm inherited the wrappers from a now defunct code base by @roboterclubaachen in 2008, where they were used to abstract a common interface for `std::thread` and whatever FreeRTOS API was available back then, so that embedded code could run on the desktop with less modification. 

I don't like these classes, I think the abstraction is very shallow and doesn't add much value. So if you want to change it, feel free to open a PR, however, it's perhaps better to use the FreeRTOS API directly and just ignore this abstraction.

kikass13, 2021-02-22:
Well.ok :)

salkinium, 2021-02-22:
I think it would make sense to implement the `std::thread`, `std::mutex` etc APIs with FreeRTOS. I'm not even sure what happens when you use `std::thread` right now in modm. It definitely won't work, the runtime is not there, but I wonder how it fails?

I'm probably also not the first person to come up with this idea, so there's likely already something out there?

closed 2021-09-29 by salkinium