modm-io/modm #570: build script generator - makes a nonsense parameter

jasa, 2021-03-02:
https://github.com/modm-io/modm/blob/ae85805096a60e8f5685f16ac31221b1484049c8/tools/build_script_generator/common.py#L221-L222

These flags are build system dependent and not filtered, will always create a useless one.   

`-ffile-prefix-map==.

salkinium, 2021-03-02:
The flags are indeed build system dependent, hence their specialization by the specific build system collector:
https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/cmake/module.lb#L48

jasa, 2021-03-03:
Should the specialization then not remove the irrelevant one instead of produce a garbage parameter? It would be the most correct  thing to do.

salkinium, 2021-03-03:
I don't understand what the issue is. These flags where added in #558 to make the file paths in the debug sections relative to the compiler and project sources. The `{gccpath}` is computed by the build system, since it differs per OS/user.

salkinium, 2021-03-03:
In CMake, it's here: https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/cmake/cmake_scripts/configure-gcc.cmake.in#L77

In SCons here: https://github.com/modm-io/modm/blob/develop/tools/build_script_generator/scons/resources/SConscript.in#L40

closed 2021-03-03 by jasa