modm-io/modm #573: general modm interrupt handling examples

kikass13, 2021-03-08:
As far as i could tell, there are no common examples for the nucleo and interrupt handling is not documented properly

## What should be there
 * Proper documentation of interrupt interface via examples
 * Proper documentation of which ISR vectors / handlers exist 
     * the same way there are typedefs for IOs and common registers via board.hpp, there should be a lookup for possible ISR table entries and their offsets 

## Examples that should be included

 * clean examples of 
     * I/O interrupt
     * timer interrupt
     * [dma interrupt (would need a DMA example, if existing)]

- - - 
Am I wrong here? What do you guys think

salkinium, 2021-03-08:
There currently is no special interrupt API other than the `NVIC_*` provided by the CMSIS core. Similiarly the [`MODM_ISR` macros are just for convenience](https://docs.modm.io/develop/api/stm32f469zit6/group__modm__architecture__interrupt.html). I think that would be the right place to add that you have to manually declare interrupts using `MODM_ISR`, unless of course the interrupts are used by the modm library (UART or I2C for example). It's currently unclear to a developer which interrupts are used by modm… There's also no protection against misspelling an interrupt handler, so the developer experience is a little miserable.

The modm-devices data does contain the interrupt table, however, it would perhaps be more useful to add the [CMSIS header Doxygen documentation](https://github.com/modm-io/cmsis-header-stm32/blob/master/stm32f0xx/Include/stm32f030x6.h) to our docs.modm.io generator.  I was under the (false) impression that it already was, but I cannot find it anywhere.

kikass13, 2021-03-08:
@salkinium at least the enum can be looked up in that header. I wasnt sure where these are defined, some simple examples should help mitigate my confusion. I successfully wrote a button example for the nucleo boards (HURRAY, im god tier now ... :P) because i was unsure about the ethernet stuff not working when using the nucleof439zi 

* Are there any plans on extending the interrupts ? 
* Should there be a wrapper around the ISR table 
    * which would be slower, but would allow to apply interrupts more generic, and also allow multiple interrupt handlers for the same interrupt source)
 * are there hardware specific things which would make such a task impossible? 
 * is this even a modm business to mess with runtime interrupt handling at all?

salkinium, 2021-03-08:
Yeah… it's complicated and that's why we haven't made any progress. We have however tried some things, see #307 and #308.

salkinium, 2021-03-08:
You can btw remap the vector table to RAM via [the `:platform:cortex-m:vector_table_location`](https://modm.io/reference/module/modm-platform-cortex-m/#vector_table_location) and the use `NVIC_SetVector` and change whatever you want. Doesn't work on Cortex-M0 though.

closed 2021-09-29 by salkinium