modm-io/modm #579: How to generate pinout table for a given device?

cocasema, 2021-03-15:
I'm looking for a way to build documentation for supported functions for board's pins in a table/matrix form.
Tried searching in the generated doxygen files, but no luck (or did I miss it?) I could only find what channels of the individual pins can be connected to, but this doesn't help much when trying to get a big picture.

```
/// Connect to Tim2 or Tim10
using Ch1 = GpioSignal;
/// Connect to Tim4
using Ch3 = GpioSignal;
```

For example, MBED has some graphical representation of the supported functions,
https://os.mbed.com/media/uploads/bcostm/nucleo_f303k8_2017_10_10.png
and a (mostly readable) source file:
https://os.mbed.com/users/mbed_official/code/mbed-dev/file/default/targets/TARGET_STM/TARGET_STM32F3/TARGET_STM32F303x8/TARGET_NUCLEO_F303K8/PeripheralPins.c/

In my case, I was particularly looking into how many and which timers are there available for use. I tried to get data into a simple table manually (and I see mbed code and modm have different opinions in some cases).

  | TIM1 | TIM2 | TIM3 | TIM4 | TIM5 | TIM6 | TIM7 | TIM8 | TIM9 | TIM10 | TIM11 | TIM12 | TIM13 | TIM14 | TIM15 | TIM16 | TIM17
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
A0 |   | ch1 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
A1 |   | ch2 |   |   |   |   |   |   |   |   |   |   |   |   | ch1n |   |  
A2 |   | ch3 |   |   |   |   |   |   |   |   |   |   |   |   | ch1 |   |  
A3 |   | ch4 |   |   |   |   |   |   |   |   |   |   |   |   | ch2 |   |  
A4 |   |   | ch2 |   |   |   |   |   |   |   |   |   |   |   |   |   |  
A5 |   | ch1 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
A6 |   |   | ch1 |   |   |   |   |   |   |   |   |   |   |   |   | ch1 |  
A7 | ch1n |   | ch2 |   |   |   |   |   |   |   |   |   |   |   |   |   | ch1
A8 | ch1 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
A9 | ch2 | ch3 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
A10 | ch3 | ch4 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
A11 | ch1n / ch4 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
A12 | ch2n |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
A13 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   | ch1 |  
A14 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   | ch1n |  
A15 |   | ch1 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
B0 | ch2n |   | ch3 |   |   |   |   |   |   |   |   |   |   |   |   |   |  
B1 | ch3n |   | ch4 |   |   |   |   |   |   |   |   |   |   |   |   |   |  
B2 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
B3 |   | ch2 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
B4 |   |   | ch1 |   |   |   |   |   |   |   |   |   |   |   |   | ch1 |  
B5 |   |   | ch2 |   |   |   |   |   |   |   |   |   |   |   |   |   | ch1
B6 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   | ch1n |  
B7 |   |   | ch4 |   |   |   |   |   |   |   |   |   |   |   |   |   | ch1n
B8 |   |   |   | ch3 |   |   |   |   |   |   |   |   |   |   |   |   |  
B9 |   | ch2 |   | ch4 |   |   |   |   |   |   | ch1 |   |   |   |   |   |  
B10 |   | ch3 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
B11 |   | ch4 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
B14 |   |   |   |   |   |   |   |   |   |   |   | ch1 |   |   |   |   |  
B15 |   |   |   |   |   |   |   |   |   |   |   | ch2 |   |   |   |   |  
  |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
F0 | ch3n |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
F1 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  

This manual process obviously doesn't scale and is error-prone.
If I want to automate it, would you suggest extending `lbuild` with more options (similar to `discover-options`),
and/or generate templates for a doxygen?

rleh, 2021-03-16:
> This manual process obviously doesn't scale

Absolutely!

> and is error-prone.

the compiler will throw error(s) if any alternate function is wrongly configured.

---

I think the right approach is to generate the missing description [here](https://github.com/modm-io/modm/blob/develop/src/modm/platform/gpio/stm32/module.lb#L145) (e.g. as a table) so that it will appear [in the documentation](https://docs.modm.io/develop/api/stm32f072vbt6/group__modm__platform__gpio.html). All necessary data are available in the file `module.md`, since this also generates the Gpio-C++-classes and `connect<>()`-methods.

The `module.description` content is parsed as markdown, see e.g. [here](https://github.com/modm-io/modm/blob/develop/src/modm/driver/usb/stusb4500.lb#L16-L27).

salkinium, 2021-03-16:
There's also `stm_layout` https://pypi.org/project/stm-layout/, which uses the modm-devices data to give you a regex-able interface to the pins and their signals.

[image]

It still doesn't give you an overview per peripheral, only per pin unfortunately. But it's still very useful.

If you're motivated, I would like to see a HTML version of this that we can generate locally for your specific device and also host on pinout.modm.io for all devices. It can

rleh, 2021-03-16:
> If you're motivated, I would like to see a HTML version of this that we can generate locally for your specific device

It could be done that way, but I would prefer to integrate that with the existing Doxypress documentation at https://docs.modm.io/. That's where most people are already looking for it anyway...

> and also host on pinout.modm.io for all devices.

I don't see the benefit of creating another domain. Why not like this (If it cannot be integrated into the Doxypress documentation or should be more interactive (Javascript or similar) than Doxypress allows):
[image]

salkinium, 2021-03-16:
Yes, of course we should integrate it into the Doxygen docs first!

I was mainly thinking of implementing a pinout picker like CubeMX, but in a standalone lightweight HTML+JS website. Including a small JS code gen for the specific modm config. As a stretch goal, code gen for other HALs could be added.

salkinium, 2021-03-16:
I mean, how difficult can it possibly be to clone this functionality? 😇 #FamousLastWords

[image]

salkinium, 2021-03-16:
Although playing around with CubeMX: It's so slow… and still surprisingly bad: There's a search field, but I cannot search for signals, only for *PINS*. WAT?!? I would want a pinout table *and* graphic that I can both search and filter.

salkinium, 2021-03-16:
@cocasema I don't have a hot fix for you now, except perhaps using `stm_layout` and CubeMX, but of course you can always check the devices datasheet (not reference manual) for such tables and use PDF search to some degree: Example [F303VCT](https://www.st.com/resource/en/datasheet/stm32f303vc.pdf#page=45)
[image]
That table but as a dynamic HTML plus the above graphic layout is basically what I would want.

cocasema, 2021-03-16:
@salkinium Thanks a lot for the useful links and info!

salkinium, 2021-03-16:
Here's a very hacky HTML generator for the above AF table: `pip3 install modm-devices jinja2`
https://gist.github.com/salkinium/43172be868a6f193897eb308a5ebe2f7
(This simple script should work fine for all devices except STM32F1 and STM32L1, which have additional signal remaps in groups).

[image]

cocasema, 2021-03-17:
Thanks a lot for the script! I was able to get the info I was looking for:

Pin | TIM1 | TIM2 | TIM3 | TIM15 | TIM16 | TIM17
-- | -- | -- | -- | -- | -- | --
A0 |   | CH1 |   |   |   |  
A1 |   | CH2 |   | CH1N |   |  
A2 |   | CH3 |   | CH1 |   |  
A3 |   | CH4 |   | CH2 |   |  
A4 |   |   | CH2 |   |   |  
A5 |   | CH1 |   |   |   |  
A6 |   |   | CH1 |   | CH1 |  
A7 | CH1N |   | CH2 |   |   | CH1
A8 | CH1 |   |   |   |   |  
A9 | CH2 | CH3 |   |   |   |  
A10 | CH3 | CH4 |   |   |   |  
A11 | CH1N, CH4 |   |   |   |   |  
A12 | CH2N |   |   |   | CH1 |  
A13 |   |   |   |   | CH1N |  
A14 |   |   |   |   |   |  
A15 |   | CH1 |   |   |   |  
B0 | CH2N |   | CH3 |   |   |  
B1 | CH3N |   | CH4 |   |   |  
B3 |   | CH2 |   |   |   |  
B4 |   |   | CH1 |   | CH1 |  
B5 |   |   | CH2 |   |   | CH1
B6 |   |   |   |   | CH1N |  
B7 |   |   | CH4 |   |   | CH1N
F0 | CH3N |   |   |   |   |  
F1 |   |   |   |   |   |

closed 2021-09-29 by salkinium